ARMmbed/mbed-os #13793: Undefined symbols using ARMC6 -O0

marcelbrucker, 2020-10-20:
### Description of defect

Using ARMC6 compiler with a slightly modified debug.json profile causes errors during linking.

`[Error] @0,0: L6218E: Undefined symbol mbed::RM1000_AT_CellularStack::SOCKET_TIMEOUT (referred from BUILD/K64F/ARMC6/mbed-os/features/cellular/framework/targets/RiotMicro/AT/RM1000_AT_CellularStack.o).`
`[Error] @0,0: L6218E: Undefined symbol mbed::UBLOX_AT_CellularStack::SOCKET_TIMEOUT (referred from BUILD/K64F/ARMC6/mbed-os/features/cellular/framework/targets/UBLOX/AT/UBLOX_AT_CellularStack.o).`
`[Error] @0,0: L6218E: Undefined symbol mbed::UBLOX_N2XX_CellularStack::SOCKET_TIMEOUT (referred from BUILD/K64F/ARMC6/mbed-os/features/cellular/framework/targets/UBLOX/N2XX/UBLOX_N2XX_CellularStack.o).`

I had to switch off compiler optimization because of weird behavior during debugging.

My used workaround:

 in `../connectivity/drivers/cellular/RiotMicro/AT/RM1000_AT_CellularStack.h`
-static constexpr seconds SOCKET_TIMEOUT = 1s;
+static const seconds SOCKET_TIMEOUT;

in `../connectivity/drivers/cellular/RiotMicro/AT/RM1000_AT_CellularStack.cpp` 
 using namespace std::chrono;

+constexpr seconds RM1000_AT_CellularStack::SOCKET_TIMEOUT = 1s;
+

... analogously for the two other undefined symbols.

#### Target(s) affected by this defect ?
FRDM-K64F

#### Toolchain(s) (name and version) displaying this defect ?
ARMC6

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.3.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
Ubuntu 20.04.1
Mbed Studio: 1.3.0

#### How is this defect reproduced ? 
1. Change "-O1" to "-O0" in the ARMC6 toolchain of debug.json profile
2. Build any program

0xc0170, 2020-10-21:
What is the Armc6 version (is the one supplied in Mbed studio correct?) ? If you do not modify debug.json, does it build?

marcelbrucker, 2020-10-23:
I don't know how to check the armc6 version. It came with Mbed Studio. If I don't modify debug.json, it successfully builds.

0xc0170, 2020-10-26:
Thanks for reporting, looks like a valid issue. We will look into it.

kjbracey-arm, 2020-10-26:
Correct fix is, I believe, to leave the header unmodified and to add

    constexpr seconds RM1000_AT_CellularStack::SOCKET_TIMEOUT;

in the cpp file.

(This is needed until C++17, which implicitly makes `static constexpr` members `inline`, so they do then not need a separate out-of-line definition).

0xc0170, 2020-10-28:
@marcelbrucker would you be able to verify this works for you and send a new pull request fixing it?

marcelbrucker, 2020-11-02:
> @marcelbrucker would you be able to verify this works for you and send a new pull request fixing it?

The solution works.
Pull Request is created - https://github.com/ARMmbed/mbed-os/pull/13843#issue-514265026.

0xc0170, 2020-11-12:
Fix merged

closed 2020-11-12 by 0xc0170