ARMmbed/mbed-os #13807: SPDX licenses in some cases cause a warning

0xc0170, 2020-10-23:
### Description of defect

The recent update to the licenses causes compiler warnings

```
[351/731] Building C object CMakeFiles/mbed-os-example-soc.../source/Security/protocols/tls_sec_prot/tls_sec_prot.c.obj ../mbed-os/connectivity/nanostack/sal-stack-nanostack/source/Security/protocols/tls_sec_prot/tls_sec_prot.c: In function 'server_tls_sec_prot_state_machine':
../mbed-os/connectivity/nanostack/sal-stack-nanostack/source/Security/protocols/tls_sec_prot/tls_sec_prot.c:510:22: warning: unused variable 'remote_eui_64' [-Wunused-variable]
  510 |             uint8_t *remote_eui_64 = sec_prot_remote_eui_64_addr_get(prot);
      |                      ^~~~~~~~~~~~~
[472/731] Building CXX object CMakeFiles/mbed-os-example-s...ty/drivers/emac/TARGET_Freescale_EMAC/kinetis_emac.cpp.obj In file included from ../mbed-os/connectivity/netsocket/./include/netsocket/EMAC.h:23,
                 from ../mbed-os/connectivity/drivers/emac/TARGET_Freescale_EMAC/kinetis_emac.h:20,
                 from ../mbed-os/connectivity/drivers/emac/TARGET_Freescale_EMAC/kinetis_emac.cpp:48:
../mbed-os/connectivity/netsocket/./include/netsocket/EMACMemoryManager.h:2:1: warning: "/*" within comment [-Wcomment]
    2 | /* SPDX-License-Identifier: Apache-2.0
      |
```

This is  the license header

```
/* Copyright (c) 2017 ARM Limited
/* SPDX-License-Identifier: Apache-2.0
 *
```

I see there is `/` extra on the second line. I saw dozen of these during the build with CMake. 

cc @harmut01 
#### Target(s) affected by this defect ?

#### Toolchain(s) (name and version) displaying this defect ?

GCC ARM at least , prints a warning

#### What version of Mbed-os are you using (tag or sha) ?

8ef0a43

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

Not related

#### How is this defect reproduced ? 

Old tools should show the same warning (using same profiles as CMake).

harmut01, 2020-10-26:
I investigated the issue and narrowed down the problem to 8 files, these were fixed in the linked pull request. Currently waiting on CI for the changes to be merged.

0xc0170, 2020-10-27:
Fixed via https://github.com/ARMmbed/mbed-os/pull/13808

PR #13808 (2020-10-27): Remove erroneous character from SPDX identifer

### Summary of changes 
Fixes #13807

Backslash was erroneously appended to comment line for SPDX identifier in some files. Removes backslash from affected files. 

#### Impact of changes 

None
#### Migration actions required
closed 2020-10-27 by 0xc0170