ARMmbed/mbed-os #13812: VLAN with Ublox C027

nadaanas1988, 2020-10-24:
Hello,

I want to ask a question about using with my Ublox-C027 with VLAN. I could't find examples that give a clear idea how to use it or confugure it. 

Please your support will be apprecieted as I'm working on really important projects and I need to meet the deadline.

Thank you in advance.

Nada

nadaanas1988, 2020-10-24:
Thank you ciarmcom  for your reply. 
I will explain my issue in details.

First of all I'm using Ublox - C027 and it will be connected in a VLAN.  My mbed will be a TCP server and my laptop will be a TCP client. I want to be able to connect and recieve other information from the client in this VLAN. My mbed was able to recieve from a TCP client without activating the VLAN ID on my laptop (client). When I add a VLAN ID to my laptop I was not able to communicate. 

I'm using mbed online compiler and as am using RTOS that mean my MBED-OS is 5.  

I need to know if I'm able to configure the VLAN on my MBED.

I hope these information are enough for explaining my issue. 

again really your help will be apprecieted.

Thank you in advance.

Nada

0xc0170, 2020-10-26:
Hi @nadaanas1988 , please use https://forums.mbed.com/ instead for questions.

I'll close this as not technical issue (this is issue tracker for Mbed OS).

closed 2020-10-26 by 0xc0170