ARMmbed/mbed-os #13823: Not able to use CMSIS-DSP from Mbed OS

MarceloSalazar, 2020-10-28:
### Description of defect

It seems CMSIS-DSP has been removed from Mbed OS with the release of Mbed OS 6.
The arm_math.h header is still present, leading to an error when linking when using a CMSIS-DSP version.

This causes problems to one of our partners trying to use this library.

```
Link: mbed-os-example-xxx
[Warning] @0,0: L3912W: Option 'legacyalign' is deprecated.
[Error] @0,0: L6218E: Undefined symbol arm_fir_f32 (referred from ...).
[Error] @0,0: L6218E: Undefined symbol arm_fir_init_f32 (referred from ...).
```

Related issues / PRs: https://github.com/ARMmbed/mbed-os/issues/12054 https://github.com/ARMmbed/mbed-os/pull/12055

#### Target(s) affected by this defect ?

S5JS100

#### Toolchain(s) (name and version) displaying this defect ?

Arm and GCC.

#### What version of Mbed-os are you using (tag or sha) ?

f57f2657f814b8c9be77782ad281c9dadcc7b2c3

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

NA

#### How is this defect reproduced ? 

@armmbed/team-samsung please confirm

@ARMmbed/mbed-os-maintainers

heuisam, 2020-10-29:
You can reproduce with s1sbp6a_example.
It is required to build mbed-os-example-s1sbp6a and mbed-dsp.
Please mbed-os use on master branch.  
https://github.com/heuisam/s1sbp6a_example
https://github.com/ARMmbed/mbed-os-example-s1sbp6a

1. You can reproduce link error if you do not import mbed-dsp.
2. You can reproduce compile error with ARM_CC if you do import mbe-dsp
3. Build success if you change tool-chain to GCC. in 2.

0xc0170, 2020-11-02:
https://github.com/ARMmbed/mbed-os/pull/13839 fixing the issue.

0xc0170, 2020-11-02:
@heuisam if you remove `arm_math.h` file from Mbed OS (what is PR above doing), are there any errors? What errors are these (you can copy-paste logs here).

heuisam, 2020-11-03:
@0xc0170  

I want to use dsp lib in mbed, so I have tried to build after importing[ mb-dsp lib.](https://os.mbed.com/users/mbed_official/code/mbed-dsp/)
It works properly with GCC. But There was compile errors in mbed studio with ARMCC as belows.
If my approch is not proper, let me know how can I use dsp APIs in mbed.

There is a log about 2nd case after remove arm_math.h
```bash
[ERROR] "./mbed-dsp/cmsis_dsp/TransformFunctions/arm_bitreversal2.S", line 69 (column 5): Error: A1137E: Unexpected characters at end of line
69 00000000     .syntax unified
^
"./mbed-dsp/cmsis_dsp/TransformFunctions/arm_bitreversal2.S", line 72 (column 2): Error: A1137E: Unexpected characters at end of line
72 00000000  .section .text
^
"./mbed-dsp/cmsis_dsp/TransformFunctions/arm_bitreversal2.S", line 73 (column 2): Error: A1137E: Unexpected characters at end of line
73 00000000  .thumb
^
"./mbed-dsp/cmsis_dsp/TransformFunctions/arm_bitreversal2.S", line 82 (column 2): Error: A1137E: Unexpected characters at end of line
82 00000000  .global arm_bitreversal_32
^
"./mbed-dsp/cmsis_dsp/TransformFunctions/arm_bitreversal2.S", line 83 (column 2): Error: A1137E: Unexpected characters at end of line
83 00000000  .global arm_bitreversal_16
^
"./mbed-dsp/cmsis_dsp/TransformFunctions/arm_bitreversal2.S", line 137: Fatal error: A1355U: A Label was found which was in no AREA
137 00000000 arm_bitreversal_32 :
6 Errors, 0 Warnings
```

If I add some preprocessor as below, there is no build errors.
``` code
 diff -r 3762170b6d4d cmsis_dsp/TransformFunctions/arm_bitreversal2.S
--- a/cmsis_dsp/TransformFunctions/arm_bitreversal2.S   Fri Nov 20 08:45:18 2015 +0000
+++ b/cmsis_dsp/TransformFunctions/arm_bitreversal2.S   Tue Oct 27 10:41:17 2020 +0900
@@ -40,7 +40,7 @@
 ;* ANY WAY OUT OF THE USE OF THIS SOFTWARE, EVEN IF ADVISED OF THE
 ;* POSSIBILITY OF SUCH DAMAGE.
 ;* -------------------------------------------------------------------- */
-#if defined(__CC_ARM)       // Keil
+#if defined(__CC_ARM) || (defined(__ARMCC_VERSION) && (__ARMCC_VERSION >= 6010050))      // Keil
     #define CODESECT AREA     ||.text||, CODE, READONLY, ALIGN=2
     #define LABEL
 #elif defined(__IASMARM__)  // IAR 
```

0xc0170, 2020-11-03:
> I want to use dsp lib in mbed, so I have tried to build after importing mb-dsp lib.

The library is very old, dated back to 2015. We meanwhile switched to ARMClang, so my first suspect (it could break). I'll add a note to that repo. If you fetch cmsis dsp from cmsis release?

0xc0170, 2020-11-03:
The latest cmsis release: https://github.com/ARM-software/CMSIS_5/releases/tag/5.7.0 (dsp is part of it)

PR #13839 (2020-11-03): cmsis: remove math header file

### Summary of changes 

It was removed in 6.0 (see reference below), and it was reintroduced when we updated
cmsis from the upstream. We missed to remove the commit adding the file in the cmsis
importer. This fixes it and the file should not be introduced again.

Fixes #13823. This is just a patch, the file should not be in post 6.0 version.

Already removed in 6.0: https://github.com/ARMmbed/mbed-os/pull/12055

#### Impact of changes
commit 0bb4ff04 (2020-11-03): cmsis: remove math header file

It was removed in 6.0 (see reference below), and it was reintroduced when we updated
cmsis from the upstream. We missed to remove the commit adding the file in the cmsis
importer. This fixes it and the file should not be introduced again.

Fixes #13823

Already removed in 6.0: https://github.com/ARMmbed/mbed-os/pull/12055
closed 2020-11-03 by 0xc0170