ARMmbed/mbed-os #13837: class "USBSerial" has no member "printf"

power-max, 2020-10-30:
### Description of defect:
The [USBSerial example](https://os.mbed.com/docs/mbed-os/v6.4/apis/usbserial.html) fails to compile due to compilation error message:
`class "USBSerial" has no member "printf"`

#### Target(s), Toolchain, and Version details:
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/WEACT_F411CE.html
PLATFORM: ST STM32 (8.1.0) > STM32F411CE (128k RAM. 512k Flash)
HARDWARE: STM32F411CE 100MHz, 128KB RAM, 512KB Flash
DEBUG: Current (blackmagic) External (blackmagic, jlink, stlink)
PACKAGES: 
 - framework-mbed 6.60200.200722 (6.2.0) 
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)

#### How is this defect reproduced ? 
Paste example code into new MBED project, attempt to compile / run (debug mode).

power-max, 2020-10-30:
I have used `printf` successfully in the past with this example using traditional (PA9 / PA10) pins following example code for bufferedSerial. However today I noticed that while this still compiles fine, no data is returned from PA9/PA10 or displayed in the Serial monitor. I believe this is because `stdout` is not writing data to the `bufferedSerial` object, `serial_port` in the case of the example. If I attempt to include the function below (as per the example) I get an `error: 'mbed::FileHandle* mbed::mbed_override_console(int)' should have been declared inside 'mbed'`

```
FileHandle *mbed::mbed_override_console(int fd)
{
    return &serial_port;
}
```

0xc0170, 2020-11-02:
cc @ARMmbed/mbed-os-core

hugueskamba, 2020-11-02:
@power-max 
Thanks for reporting this however I was not able to reproduce the problem.

The `printf()` method is [inherited](https://github.com/ARMmbed/mbed-os/blob/master/drivers/include/drivers/USBSerial.h#L51) from the [Stream class](https://github.com/ARMmbed/mbed-os/blob/master/platform/include/platform/Stream.h#L53).

I copied the code snippet from the documentation and replaced the content of https://github.com/ARMmbed/mbed-os-example-blinky/blob/development/main.cpp and built using Mbed CLI and ran `mbed compile -t GCC_ARM -m NUCLEO_F411RE`. I used `NUCLEO_F411RE` because it is a `STM32F411xE` Mbed target. 

Which Mbed target are you trying to build for? If you used Mbed CLI, can you add the command you ran to build the application?

The configuration link provided in this PR description is broken.

The only way I see your problem happening is if your configuration uses [`stdio-minimal-console-only`](https://github.com/ARMmbed/mbed-os/blob/master/platform/mbed_lib.json#L19).

0xc0170, 2020-11-02:
@power-max Closing as "can't reproduce". If this becomes valid technical issue, we can reopen.

closed 2020-11-02 by 0xc0170

power-max, 2020-11-02:
I'm using the PlatoformIO as the platform for development.

I found out later I could recreate the problem with the standard Serial
class as well (trying to call serial_object.printf caused the same error).
I saw a flag set to printf_minimal in one of the many JSON configurations I
looked at. I am fairly new to PlatformIO as a development platform so I
don't know the inner workings and the command it runs to compile and run,
least not yet.