ARMmbed/mbed-os #13838: USBSerial can't work on bluepill (stm32f1c8t6) after Mbedos 6.4.

wubaojun1982, 2020-11-01:
### Description of defect

On bluepill (stm32f103c8t6) board, the USBSerial can work on Mbedos 6.3 with #13522 #13406 files changed. But after mbedos6.4 released, it can't work now. The serial port can be identified on windows 10，but it can't work now. And there is a bug, [ERROR] .\mbed-os\targets\TARGET_STM\USBPhy_STM32.cpp:181:5: error: unknown type name 'DigitalOut'. After add "mbed.h" ，the bug is fixed. My project is configed like this:
1.My project is cloned from, https://os.mbed.com/users/hudakz/code/mbed-os-bluepill/
2. The pin map in custom PeripheralPins.c :
//*** USBDEVICE ***
MBED_WEAK const PinMap PinMap_USB_FS[] = {
{PA_11, USB_FS, STM_PIN_DATA(STM_MODE_INPUT, GPIO_NOPULL, 0)}, // USB_DM
{PA_12, USB_FS, STM_PIN_DATA(STM_MODE_INPUT, GPIO_NOPULL, 0)}, // USB_DP
{NC, NC, 0}
};
and the pin names in the custom PinNames.h
//USB pins
USB_DM = PA_11,
USB_DP = PA_12,
also added the clock source configuration in the med_app.json like this:
"target_overrides": {
"*": {
"target.clock_source": "USE_PLL_HSE_XTAL",
"target.clock_source_usb": "1",
"target.device_has_add": ["USBDEVICE"]
}
}
3. I have tested the usbserial(usbcdc) on my bluepill board with stm32cube auto code configration. It can work well.
#### Target(s) affected by this defect ?
Bluepill (STM32F1C8T6) boards

#### Toolchain(s) (name and version) displaying this defect ?
Mbed studio 1.3

#### What version of Mbed-os are you using (tag or sha) ?
Mbedos 6.4 ( 8ef0a43)

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
Mbedos 6.4 ( 8ef0a43)
#### How is this defect reproduced ? 
The bluepill(stm32f1c8t6). And there is another person @Allmoz who fixed #13406 #13522 ,also have the same bug.

Allmoz, 2020-11-01:
I confirm. Clarification, #13522 was actually is @jeromecoutant fix (thanks~).

It seems to be related to the changes made in #13683  since  `#include "mbed.h"` was removed from USBPhyHw.h, ` #include "drivers/DigitalOut.h"` is not been included nor `using namespace mbed;` is called, so the line 181 of USBPhy_STM32.cpp `DigitalOut usb_dp_pin(USB_DP, 0) ;` fails.

A solution (already tested) could be:

- add ` #include "drivers/DigitalOut.h"`  to  USBPhyHw.h
- change `DigitalOut usb_dp_pin(USB_DP, 0) ;` to `mbed::DigitalOut usb_dp_pin(USB_DP, 0) ;` in USBPhy_STM32.cpp

@jeromecoutant, @0xc0170  is this the right way? or should DigitalOut not be used there?

wubaojun1982, 2020-11-01:
After fixed digitalout bug, the serial still don't work well.

0xc0170, 2020-11-02:
@jeromecoutant Can you help fixing this ?

> After fixed digitalout bug, the serial still don't work well.

That looks like as another issue (or is it the same , 6.3 worked, 6.4 version does not work) ?

wubaojun1982, 2020-11-02:
The usb serial port can be identified，but it can't send and receive bytes.

Allmoz, 2020-11-02:
> The usb serial port can be identified，but it can't send and receive bytes.

for me this little example in bare metal works:

```
#include "mbed.h"
#include "USBSerial.h"

USBSerial usb_serial(false);

int main()
{
    ThisThread::sleep_for(5s);
    usb_serial.printf("ready\r\n");
    while (true)
    {      
        if(usb_serial.readable() && usb_serial.writeable()) {
                usb_serial.putc(usb_serial.getc());
        }
    }
}
```

give it a try

0xc0170, 2020-11-03:
@Allmoz Please review the fix https://github.com/ARMmbed/mbed-os/pull/13846

wubaojun1982, 2020-11-03:
> for me this little example in bare metal works:
>
> ```
> #include "mbed.h"
> #include "USBSerial.h"
>
> USBSerial usb_serial(false);
>
> int main()
> {
>     ThisThread::sleep_for(5s); [...]

thanks. But it still doesn't work well. Can you show me your mbed_app.json on the bare metal  mode.

wubaojun1982, 2020-11-03:
my mbed_app.json is:
{
    "requires": ["bare-metal", "drivers-usb"],
    "target_overrides": {
      "*": {
        "target.device_has_add": ["USBDEVICE"],
        "target.clock_source": "USE_PLL_HSE_XTAL",
            "target.clock_source_usb": "1"
      }
    }
}
my custom_targets.json is:
{
    "BLUEPILL": {
        "inherits": ["MCU_STM32F103x8"],
        "device_name": "STM32F103C8"
    }
}

jeromecoutant, 2020-11-03:
Don't forget 1.5KOhm pull up to USB-DP PA12

Allmoz, 2020-11-04:
> my mbed_app.json is:
> {
> "requires": ["bare-metal", "drivers-usb"],
> "target_overrides": {
> "*": {
> "target.device_has_add": ["USBDEVICE"],
> "target.clock_source": "USE_PLL_HSE_XTAL", [...]

> thanks. But it still doesn't work well. Can you show me your mbed_app.json on the bare metal mode.

And there it goes

custom_targets.json

```
{
    "STM32F103C8": {
        "inherits": [
            "MCU_STM32F103x8"
        ],
        "macros_add": [
            "STM32F103C8"
        ],
        "device_has_add": [
            "USBDEVICE"
        ],
        "device_name": "STM32F103C8"
    }
}
```

mbed_app.json

```
{
    "requires": ["bare-metal", "drivers-usb","events"],
    "macros": [ 
        "NDEBUG=1"
    ],
    "target_overrides": {
        "*": {
            "platform.stdio-baud-rate": 115200 ,
            "platform.default-serial-baud-rate" : 115200 ,
            "platform.stdio-convert-newlines"   : true,
            "platform.stdio-buffered-serial"    : true,
            "platform.stdio-flush-at-exit"      : false,
            "target.stdio_uart_tx": "PB_6",
            "target.stdio_uart_rx": "PB_7"
        }
    }
}
```
PinNames.h

```
/* mbed Microcontroller Library
 *******************************************************************************
 * Copyright (c) 2014, STMicroelectronics
 * All rights reserved.
 *
 * Redistribution and use in source and binary forms, with or without
 * modification, are permitted provided that the following conditions are met:
 *
 * 1. Redistributions of source code must retain the above copyright notice,
 *    this list of conditions and the following disclaimer.
 * 2. Redistributions in binary form must reproduce the above copyright notice,
 *    this list of conditions and the following disclaimer in the documentation
 *    and/or other materials provided with the distribution.
 * 3. Neither the name of STMicroelectronics nor the names of its contributors
 *    may be used to endorse or promote products derived from this software
[... 375 lines cut ...]
    {NC, NC, 0}
};

//*** CAN ***

MBED_WEAK const PinMap PinMap_CAN_RD[] = {
    {PA_11,      CAN_1, STM_PIN_DATA(STM_MODE_INPUT, GPIO_NOPULL, 0)},
    {PB_8,       CAN_1, STM_PIN_DATA(STM_MODE_INPUT, GPIO_NOPULL, 10)},
    {NC, NC, 0}
};

MBED_WEAK const PinMap PinMap_CAN_TD[] = {
    {PA_12,      CAN_1, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_NOPULL, 0)},
    {PB_9,       CAN_1, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_NOPULL, 10)},
    {NC, NC, 0}
};

//*** USBDEVICE ***

MBED_WEAK const PinMap PinMap_USB_FS[] = {
    {PA_11,     USB_FS, STM_PIN_DATA(STM_MODE_INPUT, GPIO_NOPULL, 0)}, // USB_DM
    {PA_12,     USB_FS, STM_PIN_DATA(STM_MODE_INPUT, GPIO_NOPULL, 0)}, // USB_DP
    {NC, NC, 0}
};
```

closed 2020-11-05 by 0xc0170

wubaojun1982, 2020-11-06:
> for me this little example in bare metal works:
>
> ```
> #include "mbed.h"
> #include "USBSerial.h"
>
> USBSerial usb_serial(false);
>
> int main()
> {
>     ThisThread::sleep_for(5s); [...]

still don't work well.

PR #13846 (2020-11-05): STM32 usb: fixing missing header file

### Summary of changes 

Fixes #13838. The previous PR #13683 did miss these unfortunately.

#### Impact of changes 

#### Migration actions required
commit 097adb57 (2020-11-05): STM32 usb: fixing missing header file

Fixes #13838