ARMmbed/mbed-os #13840: scancode:

0xc0170, 2020-11-02:
### Description of defect

The license checker scancode does not recognize our Permissive Binary License , we should add it to scancode. The quick fix until the scancode is updated, similar to what we do with SPDX, finding this license and not return an error (it is permissive).

https://github.com/ARMmbed/mbed-os/pull/13826 - this is PR affected, these files there have a problem:
```
Found files with missing license details, please review and fix
File: whd_version.h reason: Missing license header
File: 4343WA1-mfgtest_bin.c reason: Non-permissive license
File: 4343WA1-mfgtest_clm_blob.c reason: Non-permissive license
File: 4343WA1_bin.c reason: Non-permissive license
File: 4343WA1_clm_blob.c reason: Non-permissive license
File: resources.h reason: Non-permissive license
File: wifi_nvram_image.h reason: Non-permissive license
Found files with missing SPDX identifier, please review and fix
File: stm32h7_eth_init.c reason: Missing SPDX license identifier
File: PeripheralNames.h reason: Missing SPDX license identifier
File: PinNames.h reason: Missing SPDX license identifier
File: system_clock_override.c reason: Missing SPDX license identifier
File: cmsis_nvic.h reason: Missing SPDX license identifier
```

Implement similar function to `has_spdx_text_in_analysed_file` in scancode-evaluate script

@harmut01

#### Target(s) affected by this defect ?

Any

#### Toolchain(s) (name and version) displaying this defect ?

Any

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.4.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

Any

#### How is this defect reproduced ? 

Create a PR with changing any Permissive Binary License file

0xc0170, 2020-11-03:
cc @ARMmbed/mbed-os-maintainers -

0xc0170, 2020-11-03:
This should fix:

```
File: 4343WA1-mfgtest_bin.c reason: Non-permissive license
File: 4343WA1-mfgtest_clm_blob.c reason: Non-permissive license
File: 4343WA1_bin.c reason: Non-permissive license
File: 4343WA1_clm_blob.c reason: Non-permissive license
```

harmut01, 2020-11-03:
A workaround has been added to our evaluation script. I'll be briefly raising a PR to get the change approved.

harmut01, 2020-11-03:
@0xc0170 @facchinm I'd also like to raise an issue to get the licenses that are causing errors to be added to the checking software but I can't see any reference to LicenseRef-PBL in the SPDX list of licenses. Is this a valid SPDX identifier?

0xc0170, 2020-11-03:
> LicenseRef-PBL

As it is our license, we should add it. Lets talk to the license team. I've noticed also the url we provide: https://www.mbed.com/licenses/PBL-1.0 does not exist anymore. Time to fix this as well.

closed 2020-11-10 by 0xc0170

pombredanne, 2021-07-29:
Hi :wave: 
ScanCode maintainer here..
I see in https://github.com/ARMmbed/mbed-os/pull/13847#issue-514784737

> Scancode does not currently have our license in its database as PBL is not a recognized SPDX license

It is fairly easy to add these to ScanCode... just tell me what these are.

0xc0170, 2021-07-30:
@pombredanne thanks for getting back to this, we still should add the license !

The license text is https://os.mbed.com/licenses/permissive-binary-license/, the license header is:

```
/*
 * Copyright (c) 2019
 * SPDX-License-Identifier: LicenseRef-PBL
 *
 * This file and the related binary are licensed under the
 * Permissive Binary License, Version 1.0 (the "License");
 * you may not use these files except in compliance with the License.
 *
 * You may obtain a copy of the License here:
 * LICENSE-permissive-binary-license-1.0.txt and at
 * https://www.mbed.com/licenses/PBL-1.0
 *
 * See the License for the specific language governing permissions and
 * limitations under the License.
 */
```

SPDX `LicenseRef-PBL`. is there anything else required?

pombredanne, 2021-07-30:
@0xc0170 That's prefect! I created https://github.com/nexB/scancode-toolkit/issues/2617 on our side

pombredanne, 2021-07-30:
is https://www.mbed.com/licenses/PBL-1.0 the canonical URL though?

0xc0170, 2021-12-15:
> is https://www.mbed.com/licenses/PBL-1.0 the canonical URL though?

Yes, this is the url at the moment.

Thank you for fixing this!

PR #13847 (2020-11-10): Add workaround for files with permissive binary licenses

### Summary of changes 

Fixes #13840 

Add function for checking that files with Permissive Binary Licenses are not erroneously flagged as missing permissive licenses. Scancode does not currently have our license in its database as PBL is not a recognized SPDX license. This is a temporary solution until we are able to get the license into a scancode toolkit release. 
- adds function has_binary_license to determine if a file has a PBL license 
- changes condition in SPDX check branch to ensure that files with PBL are analysed, eventuall [... cut]