ARMmbed/mbed-os #13850: MUSCA_S1 targets don't compile with blinky-baremetal

MarceloSalazar, 2020-11-04:
### Description of defect

Musca targets fail to compile blinky-baremetal in Mbed OS 6

```
[mbed] Working path "/Users/.../Mbed Programs/mbed-os-example-blinky-baremetal" (program)
Building project mbed-os-example-blinky-baremetal (ARM_MUSCA_S1_NS, ARMC6)
Scan: mbed-os-example-blinky-baremetal
Configuration error: Bootloader not supported on this target. ROM start not found in targets.json.
Configuration error: Bootloader not supported on this target. RAM start not found in targets.json.
Compile [ 94.8%]: gpio_api_ns.c
[Fatal Error] gpio_api_ns.c@31,10: 'rtx_lib.h' file not found
[ERROR] ./mbed-os/targets/TARGET_ARM_SSG/TARGET_MUSCA_S1/gpio_api_ns.c:31:10: fatal error: 'rtx_lib.h' file not found
#include "rtx_lib.h"
         ^~~~~~~~~~~
1 error generated.
```

#### Target(s) affected by this defect ?

- ARM_MUSCA_S1
- ARM_MUSCA_S1_NS

GCC and ARM Errors

#### Toolchain(s) (name and version) displaying this defect ?

Arm Compiler 6 and GCC 9

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.4.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

Mbed CLI 1.10.4

#### How is this defect reproduced ? 

```
mbed import https://github.com/ARMmbed/mbed-os-example-blinky-baremetal
cd mbed-os-example-blinky
mbed compile -t <toolchain> -m <target>
```

@ARMmbed/team-arm-oss-platform

evedon, 2020-12-08:
`rtx_lib.h` is not available without an RTOS which is why this won't compile in bare metal.
It does not seem that this include is needed in `gpio_api_ns.c` so it should be removed.

evedon, 2020-12-08:
@0xc0170 Please liaise with @ARMmbed/team-arm-oss-platform to get this fixed

0xc0170, 2020-12-08:
rtx lib is used there only for finding out if interrupts are enabled. We provide critical section API to do that. It should be easy to fix this.

`core_util_is_isr_active()` instead of `IsIrqMode` 

@harmut01 @adbridge Any of you to fix this? Just take this to our board and component.

evedon, 2021-01-08:
Unfortunately this is not the only issue.

There are many dependencies on TFM platform which need to be sorted (i.e. removed)
```
Building project mbed-os-example-blinky-baremetal (ARM_MUSCA_S1, GCC_ARM)
Scan: mbed-os-example-blinky-baremetal
Configuration error: Bootloader not supported on this target. ROM start not found in targets.json.
Configuration error: Bootloader not supported on this target. RAM start not found in targets.json.
Compile [ 93.3%]: gpio_api_ns.c
[Fatal Error] gpio_api_ns.c@32,10: tfm_platform_api.h: No such file or directory
[ERROR] ./mbed-os/targets/TARGET_ARM_SSG/TARGET_MUSCA_S1/gpio_api_ns.c:32:10: fatal error: tfm_platform_api.h: No such file or directory
   32 | #include "tfm_platform_api.h"
      |          ^~~~~~~~~~~~~~~~~~~~
compilation terminated.
```

Assigning to @gbrtth

0xc0170, 2021-05-12:
@gbrtth Is this still valid issue?

gbrtth, 2021-05-12:
> 
> 
> @gbrtth Is this still valid issue?

Yes, the issue is still present.

closed 2021-10-20 by ciarmcom