ARMmbed/mbed-os #13852: SFE_* targets don't compile with blinky-baremetal

MarceloSalazar, 2020-11-04:
### Description of defect

S1SBP6A target fails to compile blinky-baremetal in Mbed OS 6

Arm and GCC Error

```
[mbed] Working path "/Users/.../Mbed Programs/mbed-os-example-blinky-baremetal" (program)
Building project mbed-os-example-blinky-baremetal (SFE_ARTEMIS, GCC_ARM)
Scan: mbed-os-example-blinky-baremetal
Configuration error: Bootloader not supported on this target. ROM start not found in targets.json.
Configuration error: Bootloader not supported on this target. RAM start not found in targets.json.
Compile [  5.8%]: hci_drv_apollo3.c
[Fatal Error] hci_drv_apollo3.c@52,10: wsf_types.h: No such file or directory
[ERROR] ./mbed-os/connectivity/drivers/ble/FEATURE_BLE/TARGET_Ambiq_Micro/hal/apollo3/hci_drv_apollo3.c:52:10: fatal error: wsf_types.h: No such file or directory
   52 | #include "wsf_types.h"
      |          ^~~~~~~~~~~~~
compilation terminated.
```

#### Target(s) affected by this defect ?

SFE_ 

#### Toolchain(s) (name and version) displaying this defect ?

Arm Compiler 6 and GCC 9

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.4.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

Mbed CLI 1.10.4

#### How is this defect reproduced ? 

```
mbed import https://github.com/ARMmbed/mbed-os-example-blinky-baremetal
cd mbed-os-example-blinky
mbed compile -t <toolchain> -m <target>
```

@Wenn0101

Wenn0101, 2020-11-05:
Thanks for the report. We will take a look.

0xc0170, 2020-11-12:
> Thanks for the report. We will take a look.
 
Any update ?

PR #14460 (2021-03-31): Ambiq: Fix baremetal build profile when using CLI 1

### Summary of changes 

Fixes #13852

Add mbed_lib.json in the Ambiq target folder of FEATURE_BLE to make sure
the library is excluded in the baremetal build profile unless explicitly
added in `mbed_app.json`.

#### Impact of changes 
None 

#### Migration actions required 
None
closed 2021-03-31 by adbridge