ARMmbed/mbed-os #13854: Error when mbed-os-example-wifi build using Musca-S1

GiungKim, 2020-11-04:
### Description of defect

I'm trying to apply the example of mbed-os-example-wifi using Musca-S1, but an error occurs as follows.
I have installed the 'cbor' Python module, but the same error appear.
Is there any way to solve this error?
[image]

#### Target(s) affected by this defect ?

* Arm V2M-Musca-S1

#### Toolchain(s) (name and version) displaying this defect ?

* Arm Compiler 6.14

#### What version of Mbed-os are you using (tag or sha) ?

* mbed-os 6.4.0
  Commit: 8ef0a435b2356f8159dea8e427b2935d177309f8
  Imported from: https://github.com/ARMmbed/mbed-os/

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

* Mbed Studio: 1.3.0

#### How is this defect reproduced ? 

1. Click "File -> New Program" and create "MBED OS 6 mbed-os-example-wifi"
[image]

2. Update from mbed-os 6.3.0 to  mbed-os 6.4.0
[image]

3. Set target
[image]

4. Build program
[image]

5. Error
[image]

0xc0170, 2020-11-04:
I assume mbed studio uses virtual env. @arekzaluski is that correct? How a user installs additional python modules?

arekzaluski, 2020-11-04:
@GiungKim  Mbed Studio is using an isolated version of Python. You can install new dependencies by:

1. Opening a new terminal: `Menu->Terminal-New Terminal`
2. Install cbor using pip: `pip install cbor`
3. Click build

It should build correctly now. We will add `cbor` dependency to a new version of Studio.

GiungKim, 2020-11-05:
Thanks for your help.

I solved the problem.

closed 2020-11-06 by 0xc0170