ARMmbed/mbed-os #13857: nRF52 target doesn't compile with blinky-baremetal

MarceloSalazar, 2020-11-04:
### Description of defect

nrf52_dk target fails to compile/link blinky-baremetal in Mbed OS 6

Arm error (GCC works ok)

```
Link: mbed-os-example-blinky-baremetal
[Warning] @0,0: L3912W: Option 'legacyalign' is deprecated.
[Error] @0,0: L6218E: Undefined symbol Image$$ARM_LIB_HEAP$$ZI$$Base (referred from BUILD/NRF52_DK/ARM/mbed-os/platform/source/mbed_sdk_boot.o).
[Error] @0,0: L6218E: Undefined symbol Image$$ARM_LIB_HEAP$$ZI$$Length (referred from BUILD/NRF52_DK/ARM/mbed-os/platform/source/mbed_sdk_boot.o).
Warning: L3912W: Option 'legacyalign' is deprecated.
Error: L6218E: Undefined symbol Image$$ARM_LIB_HEAP$$ZI$$Base (referred from BUILD/NRF52_DK/ARM/mbed-os/platform/source/mbed_sdk_boot.o).
Error: L6218E: Undefined symbol Image$$ARM_LIB_HEAP$$ZI$$Length (referred from BUILD/NRF52_DK/ARM/mbed-os/platform/source/mbed_sdk_boot.o).
Finished: 0 information, 1 warning and 2 error messages.
[ERROR] Warning: L3912W: Option 'legacyalign' is deprecated.
Error: L6218E: Undefined symbol Image$$ARM_LIB_HEAP$$ZI$$Base (referred from BUILD/NRF52_DK/ARM/mbed-os/platform/source/mbed_sdk_boot.o).
Error: L6218E: Undefined symbol Image$$ARM_LIB_HEAP$$ZI$$Length (referred from BUILD/NRF52_DK/ARM/mbed-os/platform/source/mbed_sdk_boot.o).
Finished: 0 information, 1 warning and 2 error messages.
```

#### Target(s) affected by this defect ?

nrf52_dk 

#### Toolchain(s) (name and version) displaying this defect ?

Arm Compiler 6

#### What version of Mbed-os are you using (tag or sha) ?

cdc2d45e05622f375d222b1920a9df148c5acbab

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

Mbed CLI 1.10.4

#### How is this defect reproduced ? 

```
mbed import https://github.com/ARMmbed/mbed-os-example-blinky-baremetal
cd mbed-os-example-blinky
mbed compile -t <toolchain> -m <target>
```

@ARMmbed/mbed-os-maintainers

LDong-Arm, 2020-11-23:
PR created to add a heap region and support bare metal: https://github.com/ARMmbed/mbed-os/pull/13951

yogeshk19, 2020-11-24:
This doesn't compile for NRF52840 for GCC as well. Is this going to be fixed all NRF52 BLE modules?

Thanks,
Yogesh

LDong-Arm, 2020-11-25:
@yogeshk19 As far as I tested, the blinky-baremetal example compiles for NRF52840_DK for all versions from mbed-os-6.0.0 onwards. It doesn't compile (some cryptocell dependency issue) for 5.15.

PR #13951 (2020-11-26): MCU_NRF52832 (NRF52_DK, SDT52832B): use two-region memory model to support MicroLib

### Summary of changes 

Fixes:  #13857 

Arm toolchain's MicroLib, a lightweight C library used in conjunction with the bare metal profile to save code size, requires two separate memory regions for stack and heap. For MCU_NRF52832 (NRF52_DK, SDT52832B) such memory model is already used in [the GCC_ARM linker script](https://github.com/ARMmbed/mbed-os/blob/mbed-os-6.5.0/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_NRF52/TARGET_MCU_NRF52832/device/TOOLCHAIN_GCC_ARM/NRF52832.ld), and this PR adds it to the Arm Com [... cut]
closed 2020-11-26 by 0xc0170