ARMmbed/mbed-os #13858: Link error when using cmake and L475 + small CLIB

MarceloSalazar, 2020-11-04:
### Description of defect

Link error when trying to use the small CLIB with cmake

```
armclang: error: no such file or directory: '"--library_type=microlib"'
[16/260] Building C object CMakeFiles\mbed-os-example-blinky.dir\mbed-os\cmsis\CMSIS_5\CMSIS\RTOS2\RTX\Source\rtx_kernel.o
ninja: build stopped: subcommand failed.
```

#### Target(s) affected by this defect ?

DISCO L475 

#### Toolchain(s) (name and version) displaying this defect ?

Arm Compiler 6

#### What version of Mbed-os are you using (tag or sha) ?

feature-cmake

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

NA

#### How is this defect reproduced ? 

- Import mbed-os-example-blinky
- Switch app and Mbed OS to feature-cmake
- Configure: `mbedtools configure -m DISCO_L475VG_IOT01A -t ARM`
- Update \.mbedbuild\mbed_config.cmake and enable the small CLIB instead of std: `set(MBED_C_LIB "small" CACHE STRING "")`
- Compile `mkdir build && cd build && cmake .. -GNinja && cmake --build .`

@0xc0170 @hugueskamba

0xc0170, 2020-11-04:
We will look at this

hugueskamba, 2020-11-05:
@MarceloSalazar 
I am looking at it but please note that you cannot build `mbed-os-example-blinky` with the small library because Microlib (and newlib-nano) are not thread-safe and the application uses threads.

MarceloSalazar, 2020-11-05:
OK, can we try with Blinky baremetal?
I'd like to make sure the right cmake config files are in place for us to use the minimal clib. Thanks.

hugueskamba, 2020-11-05:
> OK, can we try with Blinky baremetal?
> I'd like to make sure the right cmake config files are in place for us to use the minimal clib. Thanks.

Blinky baremetal work is in progress. Regardless of that, Blinky should have at least built as the option used to select Microlib is [correct as per documentation](https://developer.arm.com/documentation/dui0808/e/the-arm-c-micro-library/building-an-application-with-microlib) and I do remember testing it when it was implemented.

closed 2020-11-06 by hugueskamba