ARMmbed/mbed-os #13873: Error when mbed-os-example-pelion build using NUCLEO-F411RE

GiungKim, 2020-11-06:
### Description of defect

I'm trying to work the example "mbed-os-example-pelion" using NUCLEO-F411RE, but an error occurs as below.
Is there way to solve this error?

[image]

[image]

#### Target(s) affected by this defect ?

* NUCLEO-F411RE

#### Toolchain(s) (name and version) displaying this defect ?

* Arm Compiler 6.14

#### What version of Mbed-os are you using (tag or sha) ?

* mbed-os 6.4.0
  Commit : 8ef0a43
  Imported from : https://github.com/ARMmbed/mbed-os.git

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

* Mbed Studio : 1.3.0

#### How is this defect reproduced ? 

1. Click "File -> New Program" and create "MBED OS 5 mbed-os-example-pelion"

[image]

2. Update from libraries

[image]

3. Set target

[image]

4. Build program

[image]

5. Error

[image]

teetak01, 2020-11-06:
Hi @GiungKim this issue is due to PSA API changes in Mbed OS 6.x. Next Device Management Client release will support this board without dependencies to PSA APIs (which are experimental in 6.x).

Meanwhile I would recommend to continue to use the Mbed OS version that is defined in the application repository (5.15.3).

Sorry for the inconvenience.

0xc0170, 2020-11-06:
I'll close this as resolved. Thanks @teetak01

closed 2020-11-06 by 0xc0170