ARMmbed/mbed-os #13875: S1SBP6A fails RTC Greentea tests

MarceloSalazar, 2020-11-06:
### Description of defect

The S1SBP6A target fails to run the RTC test with both Arm and GCC compilers.
This is a MUST [requirement](https://os.mbed.com/mbed-enabled/requirements/boards/) for Mbed Enabled.

```
| target          | platform_name | test suite                                     | test case                                                       | passed | failed | result  | elapsed_time (sec) |
|-----------------|---------------|------------------------------------------------|-----------------------------------------------------------------|--------|--------|---------|--------------------|
| S1SBP6A-GCC_ARM | S1SBP6A       | mbed-os-hal-tests-tests-mbed_hal-rtc           | RTC - accuracy                                                  | 0      | 0      | SKIPPED | 0.0                |
| S1SBP6A-GCC_ARM | S1SBP6A       | mbed-os-hal-tests-tests-mbed_hal-rtc           | RTC - enabled                                                   | 0      | 0      | SKIPPED | 0.0                |
| S1SBP6A-GCC_ARM | S1SBP6A       | mbed-os-hal-tests-tests-mbed_hal-rtc           | RTC - glitch                                                    | 1      | 0      | OK      | 3.73               |
| S1SBP6A-GCC_ARM | S1SBP6A       | mbed-os-hal-tests-tests-mbed_hal-rtc           | RTC - init                                                      | 1      | 0      | OK      | 0.03               |
| S1SBP6A-GCC_ARM | S1SBP6A       | mbed-os-hal-tests-tests-mbed_hal-rtc           | RTC - persist                                                   | 1      | 0      | OK      | 4.04               |
| S1SBP6A-GCC_ARM | S1SBP6A       | mbed-os-hal-tests-tests-mbed_hal-rtc           | RTC - range                                                     | 0      | 1      | FAIL    | 8.11               |
| S1SBP6A-GCC_ARM | S1SBP6A       | mbed-os-hal-tests-tests-mbed_hal-rtc           | RTC - write/read                                                | 0      | 0      | SKIPPED | 0.0                |
| S1SBP6A-GCC_ARM | S1SBP6A       | mbed-os-hal-tests-tests-mbed_hal-rtc_reset     | mbed-os-hal-tests-tests-mbed_hal-rtc_reset                      | 1      | 0      | OK      | 29.16              |
| S1SBP6A-GCC_ARM | S1SBP6A       | mbed-os-hal-tests-tests-mbed_hal-rtc_time      | test is leap year - RTC leap years full support                 | 1      | 0      | OK      | 0.08               |
| S1SBP6A-GCC_ARM | S1SBP6A       | mbed-os-hal-tests-tests-mbed_hal-rtc_time      | test is leap year - RTC leap years partial support              | 1      | 0      | OK      | 0.08               |
| S1SBP6A-GCC_ARM | S1SBP6A       | mbed-os-hal-tests-tests-mbed_hal-rtc_time      | test local time - invalid param                                 | 1      | 0      | OK      | 0.06               |
| S1SBP6A-GCC_ARM | S1SBP6A       | mbed-os-hal-tests-tests-mbed_hal-rtc_time      | test make time - invalid param                                  | 1      | 0      | OK      | 0.06               |
| S1SBP6A-GCC_ARM | S1SBP6A       | mbed-os-hal-tests-tests-mbed_hal-rtc_time      | test make time boundary values - RTC leap years full support    | 1      | 0      | OK      | 0.09               |
| S1SBP6A-GCC_ARM | S1SBP6A       | mbed-os-hal-tests-tests-mbed_hal-rtc_time      | test make time boundary values - RTC leap years partial support | 1      | 0      | OK      | 0.1                |
| S1SBP6A-GCC_ARM | S1SBP6A       | mbed-os-hal-tests-tests-mbed_hal-rtc_time      | test set_time twice                                             | 1      | 0      | OK      | 2.03               |
| S1SBP6A-GCC_ARM | S1SBP6A       | mbed-os-hal-tests-tests-mbed_hal-rtc_time_conv | test make time and local time - RTC leap years full support     | 1      | 0      | OK      | 9.24               |
| S1SBP6A-GCC_ARM | S1SBP6A       | mbed-os-hal-tests-tests-mbed_hal-rtc_time_conv | test make time and local time - RTC leap years partial support  | 1      | 0      | OK      | 9.24               |
mbedgt: test case results: 3 SKIPPED / 13 OK / 1 FAIL

```

#### Target(s) affected by this defect ?

S1SBP6A

#### Toolchain(s) (name and version) displaying this defect ?

ARM, GCC_ARM (official versions)

#### What version of Mbed-os are you using (tag or sha) ?

77638f9b725592629b38b9124e7b56f72cc75401

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

1.10.4

#### How is this defect reproduced ? 

```
mbed test -t GCC_ARM -m S1SBP6A -v -n *mbed-os-hal-tests-tests-mbed_hal-rtc*
mbed test -t ARM -m S1SBP6A -v -n *mbed-os-hal-tests-tests-mbed_hal-rtc*
```

@ARMmbed/team-samsung

MarceloSalazar, 2020-11-06:
Closing as duplicate of https://github.com/ARMmbed/mbed-os/issues/13876

closed 2020-11-06 by MarceloSalazar

heuisam, 2020-11-08:
@MarceloSalazar ,

It was known issue, I already ask for this issue to @toyowata.
https://github.com/ARMmbed/Samsung_BP6A_project/issues/5#issue-622289207

> RTC register of BP6A is not cover from 0 to xEFFFFFFF.
> It can cover 0~0x0FFFFFFF.
> mbed_hal_rtc tc was passed after applying the below patch.
> please check the tc.

```
diff --git a/TESTS/mbed_hal/rtc/main.cpp b/TESTS/mbed_hal/rtc/main.cpp
index 328a051f69..cf0e43c117 100644
--- a/TESTS/mbed_hal/rtc/main.cpp
+++ b/TESTS/mbed_hal/rtc/main.cpp
@@ -158,7 +158,7 @@ void rtc_range_test()
 {
     static const uint32_t starts[] = {
         0x00000000,
-        0xEFFFFFFF,
+        0x0FFFFFFF,
         0x00001000,
         0x00010000,
     };
(END)
```