ARMmbed/mbed-os #13884: M2354 fails to TRNG Greentea test

MarceloSalazar, 2020-11-09:
### Description of defect

The M2354 target fails to pass the TRNG test on the 5.15 branch.

```
| target                   | platform_name | test suite                  | test case       | passed | failed | result | elapsed_time (sec) |
|--------------------------|---------------|-----------------------------|-----------------|--------|--------|--------|--------------------|
| NU_M2354_NPSA_NS-GCC_ARM | NU_M2354      | mbed-os-tests-mbed_hal-trng | TRNG: trng_test | 0      | 0      | ERROR  | 0.0                |
mbedgt: test case results: 1 ERROR
```

#### Target(s) affected by this defect ?

NU_M2354_NPSA_NS

#### Toolchain(s) (name and version) displaying this defect ?

Official Arm and GCC version

#### What version of Mbed-os are you using (tag or sha) ?

b114a9c878519d6489ac3426697196bbea34c8ea

5.15 branch

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

1.10.4

#### How is this defect reproduced ? 

```
mbed test -t GCC_ARM -m NU_M2354_NPSA_NS -v -n *mbed-os-tests-mbed_hal-trng*
mbed test -t ARM -m NU_M2354_NPSA_NS -v -n *mbed-os-tests-mbed_hal-trng*
```

@ARMmbed/team-nuvoton

0xc0170, 2020-11-12:
@ARMmbed/team-nuvoton Please review

cyliangtw, 2020-11-12:
After discussion with Marcelo, it's only failed on Mac/OS. Nuvoton also reproduced locally and identified **serial break reset** failed on Mac, so far we notified Nu-Link team to fix it.

Snippet of the good test result on Windows:
```
mbedgt: mbed-host-test-runner: stopped and returned 'OK'
mbedgt: test on hardware with target id: 13120000202008190123134300000001
mbedgt: test suite 'tests-mbed_hal-trng' ............................................................. OK in 19.41 sec
        test case: 'TRNG: trng_test' ................................................................. OK in 0.58 sec
mbedgt: test case summary: 1 pass, 0 failures
mbedgt: all tests finished!
mbedgt: shuffle seed: 0.7824117791
mbedgt: test suite report:
| target                 | platform_name | test suite          | result | elapsed_time (sec) | copy_method |
|------------------------|---------------|---------------------|--------|--------------------|-------------|
| NU_M2354_NPSA_NS-ARMC6 | NU_M2354      | tests-mbed_hal-trng | OK     | 19.41              | default     |
mbedgt: test suite results: 1 OK
mbedgt: test case report:
| target                 | platform_name | test suite          | test case       | passed | failed | result | elapsed_time (sec) |
|------------------------|---------------|---------------------|-----------------|--------|--------|--------|--------------------|
| NU_M2354_NPSA_NS-ARMC6 | NU_M2354      | tests-mbed_hal-trng | TRNG: trng_test | 1      | 0      | OK     | 0.58               |
mbedgt: test case results: 1 OK
mbedgt: completed in 20.18 sec

```

MarceloSalazar, 2020-11-12:
Thanks @cyliangtw 
We shall close this issue here as the actual RTC test pass on Windows.
We'll track of this known issue in the platforms' page.

closed 2020-11-12 by MarceloSalazar