ARMmbed/mbed-os #13888: STM32G070 compile error

caoddx, 2020-11-10:
### Description of defect

`STM32G070xx` does not seem to have `TIM2` :

`[Error] us_ticker_data.h@26,22: 'TIM2' undeclared`

And has `USART3_4_IRQn` without `USART3_4_LPUART1_IRQn` :

`[Error] serial_device.c@132,17: 'USART3_4_LPUART1_IRQn' undeclared (first use in this function)`
`[Error] serial_device.c@329,21: 'USART3_4_LPUART1_IRQn' undeclared (first use in this function)`

#### Target(s) affected by this defect ?

Custom target inherits from `MCU_STM32G070xx`

#### Toolchain(s) (name and version) displaying this defect ?

`GCC_ARM`

#### What version of Mbed-os are you using (tag or sha) ?

`mbed-os-6.4.0`

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

`mbed-cli`

0xc0170, 2020-11-10:
> We cannot automatically identify a release based on the version of Mbed OS that you have provided.
Please provide either a single valid sha of the form #abcde12 or #3b8265d70af32261311a06e423ca33434d8d80de
or a single valid release tag of the form mbed-os-x.y.z .
E.g. 'mbed-os-6.4.0' has not been matched as a valid tag or sha.How can we reproduce your issue?

@ARMmbed/mbed-os-maintainers this looks valid release to me, or why our script failed?

0xc0170, 2020-11-10:
cc @ARMmbed/team-st-mcd

jeromecoutant, 2020-11-13:
Hi
I confirm the issue.
Patch is coming soon

PR #13906 (2020-11-27): STM32G070: generic target creation
### Summary of changes 

MCU_STM32G070xx is created in targets.json file
in order to ease custom boards creation.

Few compilation issue solved for that uses case.

Fix #13888 

#### Impact of changes 

#### Migration actions required 

### Documentation 

----------------------------------------------------------------------------------------------------------------
### Pull request type
closed 2020-11-27 by 0xc0170