ARMmbed/mbed-os #13903: Disabling extended advertising makes BLE connections fail after the first connection

jrobeson, 2020-11-12:
### Description of defect
Disabling BLE Extended Advertising makes BLE connections fail after the first connection

Here's the setting I used with the [Heartrate Example](https://github.com/ARMmbed/mbed-os-example-ble/tree/master/BLE_HeartRate)

```json
  "target_overrides": {
    "*": {
      "ble.ble-feature-extended-advertising": false,
    }
```

#### Target(s) affected by this defect ?
I don't have an official target, but here's my custom one https://gist.github.com/jrobeson/0c5130aebb8017e950026fe244a5bd98

It's barely different than any other nrf52832. It could also easily affect any other BLE target or at least nordic specific ones. It's just that the nrf52832 does not support extended advertising, so i wanted to turn it off.

#### Toolchain(s) (name and version) displaying this defect ?
arm-none-eabi-gcc --version                                                                                                      2020-11-12 03:31:02
arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 9-2019-q4-major) 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

#### What version of Mbed-os are you using (tag or sha) ?
mbed-os-99.99.99
mbed-os-6.4.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed --version                                                                                                                                            2020-11-12 03:39:23
1.10.4

#### How is this defect reproduced ? 

See Above

jrobeson, 2020-11-12:
It seems like extended advertising should have been off by default for nrf52832 devices any other BLE device that doesn't support extended advertising.

paul-szczepanek-arm, 2021-03-18:
One important thing to ask is: are you restarting advertising after disconnection? After a device connects advertising stops. You have to restart advertising yourself.
For now I assume you do but please check. Please try on an up to date example.

In my investigation I uncovered a few things. The example works OK, whether extended advertising feature is enabled or disabled. (I discovered a problem with the payload https://github.com/ARMmbed/mbed-os-example-ble/pull/361/files but that is wholly unrelated to this issue).

There is however a problem with nrf tools. Nrf tools will often refuse to connect a second time but that is nothing to do with advertising and some internal problem with nrf tools. If you start connecting on the device tab and it's not connecting, go back to the scanning tab and initiate a scan - nrf tools will then connect immediately.

pan-, 2021-03-24:
@jrobeson The issue seems to be resolved. Please reopen if you're not satisfied with the resolution.

closed 2021-03-24 by pan-