ARMmbed/mbed-os #13916: SetRead/WriteAuthorizationCallbacks ignored if called after containing service is added to the GattServer

AGlass0fMilk, 2020-11-18:
### Description of defect

Attempting to set the read/write authorization callbacks for a `GattCharacteristic` after the enclosing service has already been added to the `GattServer` results in the authorization callbacks never being installed.

This behavior may be acceptable but it is nowhere mentioned as a caveat in the doc comments.

#### Target(s) affected by this defect ?

NRF52840_DK

#### Toolchain(s) (name and version) displaying this defect ?
GCC

```
arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 9-2019-q4-major) 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

#### What version of Mbed-os are you using (tag or sha) ?

05489818e9f6eeebb87233232d77374983e18142

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
1.10.4

#### How is this defect reproduced ? 
Attempt the following code when setting up a service:

```
    _server = &ble_interface.gattServer();
    _server->addService(_dfu_service);

    /** Setup write authorization callbacks */
    _slot_char.setWriteAuthorizationCallback(this, &DFUService::on_slot_write_request);
    _offset_char.setWriteAuthorizationCallback(this, &DFUService::on_offset_write_request);
    _dfu_ctrl_char.setWriteAuthorizationCallback(this, &DFUService::on_dfu_ctrl_write_request);
```

The authorization callbacks will never be executed, which is not expected behavior based on the documentation.

pan-, 2020-11-23:
@AGlass0fMilk Thanks for raising the issue, most of the operations are not taken into account if the characteristic has been registered into the server. We should amend the documentation to reflect this.

AGlass0fMilk, 2020-12-15:
> @AGlass0fMilk Thanks for raising the issue, most of the operations are not taken into account if the characteristic has been registered into the server. We should amend the documentation to reflect this.

Something else that has come up in the experimental-ble-services repository is what the required lifetime of the characteristic/service objects passed into the BLE stack is.

The documentation should clarify how exactly the parameters are stored in the stack (if at all). This will help avoid issues like this -- if the service object is already registered and copied internally to the stack then obviously changes to the original object will not be propagated to the stack's copy.