ARMmbed/mbed-os #13923: 6.x CAN support for STM32

tdjastrzebski, 2020-11-18:
mbed-os 6.x by default does not use new HAL CAN module, legacy version is being used instead.
* stm32l4xx_hal_conf.h
* stm32f7xx_hal_conf.h
* stm32f4xx_hal_conf.h
* stm32f1xx_hal_conf.h

The above files contain the following two lines:
``` C++
// #define HAL_CAN_MODULE_ENABLED
#define HAL_CAN_LEGACY_MODULE_ENABLED
```
Probably legacy CAN is enabled because `can_api.c` depends on it.
However, this prevents usage of new code examples.
Maybe it would be a good idea to update `can_api.c`

jeromecoutant, 2020-11-19:
> Maybe it would be a good idea to update `can_api.c`

Yes, of course :-)

0xc0170, 2020-11-23:
@tdjastrzebski Please keep the issue template (we are parsing the template to identify the issues and better triage them). Thanks.

MubeenHCLite, 2021-07-01:
A PR is raised with IP definition updated to current HAL implementation

jeromecoutant, 2021-07-07:
> A PR is raised with IP definition updated to current HAL implementation

#14846