ARMmbed/mbed-os #13926: STM32F207ZG I2CSlave issue

KoolePV, 2020-11-19:
### Description of defect
The i2cslave (example code of documentation) with this target can only be used when there is no sleep in the same thread and no other threads are running. When a sleep is added in the thread, the software stalls after the first time the slave is called by the master. Placing a sleep is not a problem when the F746ZG is used. 

#### Target(s) affected by this defect ?
STM32F207ZG

#### Toolchain(s) (name and version) displaying this defect ?
NA

#### What version of Mbed-os are you using (tag or sha) ?
I tried this with mbed-os-6.0.0 till mbed-os-6.4.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
NA

#### How is this defect reproduced ? 
When the standard example code of the i2cslave is compiled, the i2cslave works fine. But when a sleep is added ThisThread::sleep_for(), the software does not return from the sleep.

0xc0170, 2020-11-23:
cc @ARMmbed/team-st-mcd Please review

jeromecoutant, 2020-11-23:
@KoolePV Could you try again after removing 

https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32F2/TARGET_NUCLEO_F207ZG/device/system_clock.c#L151-L154

from code ?

KoolePV, 2020-11-25:
> @KoolePV Could you try again after removing
> 
> https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32F2/TARGET_NUCLEO_F207ZG/device/system_clock.c#L151-L154
> 
> from code ?

I removed this part and the problem still exists.

jeromecoutant, 2020-11-27:
I quickly tried, it didn't work.... :-(
Could you attach main.cpp file for both master and slave ?
Thx

KoolePV, 2020-12-04:
@jeromecoutant 
[I2C.zip](https://github.com/ARMmbed/mbed-os/files/5643266/I2C.zip)

KoolePV, 2020-12-23:
Does somebody have some ideas that I could try to resolve this problem?