ARMmbed/mbed-os #13933: [BLE] Service Changed characteristic CCCD not available

AGlass0fMilk, 2020-11-19:
### Description of defect

According to the Bluetooth Core Specification 5.2, Volume 3, Part G, Section 7.1 (Service Changed) the _Service Changed Characteristic_ only supports providing information to clients through the _indication_ mechanism. See excerpt below:

[image]

In Mbed's BLE stack, this characteristic is intrinsically created in the Cordio `GattServerImpl` implementation. Nowhere is it available through the standard Mbed BLE API to allow the user to indicate to a client that the service structure of the device has changed:

https://github.com/ARMmbed/mbed-os/blob/f2278567d09b9ae9f4843e1d9d393526b9462783/connectivity/FEATURE_BLE/source/cordio/source/GattServerImpl.cpp#L1246-L1312

**The actual issue here is:**

I have enumerated all the characteristics and services on a basic Mbed BLE device using several tools, including nRF Connect with an attached nRF52840 Dongle, as well as a custom python script using my laptop's internal bluetooth chipset.

The issue I have found is that the CCCD for the implicit Service Changed Characteristic is not actually available to the connected client for some reason! This is clearly a violation of the Bluetooth Core Specification!

[image]
bleak python script showing no CCCD on the implicitly-created Service Changed Characteristic

[image]
nRF Connect script showing no CCCD on the implicitly-created Service Changed Characteristic

I have had long-standing issues pairing/bonding with mobile OS's from an Mbed device, both Android _and_ iOS. 

When I attempt to pair with a more recent Android version the connection is actually terminated! I'm thinking this issue may be related. Android likely attempts to enable indications for the service by writing the CCCD after pairing/bonding. Since the CCCD is unavailable, Android's BLE stack blows up (this is just a theory right now).

#### Target(s) affected by this defect ?
All BLE devices using Cordio stack

#### Toolchain(s) (name and version) displaying this defect ?
(GNU Tools for Arm Embedded Processors 9-2019-q4-major) 2.33.1.20191025

#### What version of Mbed-os are you using (tag or sha) ?

05489818e9f6eeebb87233232d77374983e18142

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
```
(python3-mbed) gdbeckstein@magic-man:~/Documents/ble-dfu/mbed-os$ pip list
Package             Version     Location
------------------- ----------- -------------------------------------------------------------------
appdirs             1.4.3
asn1ate             0.6.0
attrs               19.3.0
Automat             20.2.0
beautifulsoup4      4.6.3
ble-serial          1.2.1
bleak               0.9.1
bluepy              1.3.0
cbor                1.0.0
certifi             2019.11.28
cffi                1.14.1
chardet             3.0.4
Click               7.0
cmsis-pack-manager  0.2.10
cobs                1.1.4
colorama            0.3.9
constantly          15.1.0
crc16               0.1.1
crccheck            0.6
cryptography        2.9.2
cycler              0.10.0
docopt              0.6.2
ecdsa               0.15
elftools            0.1.0.dev0
fasteners           0.15
Flask               1.1.2
future              0.16.0
futures             3.1.1
fuzzywuzzy          0.18.0
grip                4.5.2
hidapi              0.9.0.post2
hyperlink           20.0.1
icetea              1.2.4
idna                2.7
imgtool             1.7.0rc1
importlib-metadata  1.6.0
incremental         17.5.0
iniconfig           1.1.1
intelhex            2.2.1
itsdangerous        1.1.0
Jinja2              2.10.3
jsonmerge           1.7.0
jsonschema          2.6.0
junit-xml           1.8
kiwisolver          1.2.0
lockfile            0.12.2
Logbook             1.5.3
manifest-tool       1.5.2
Markdown            3.3.2
MarkupSafe          1.1.1
matplotlib          3.3.0
mbed-ble-test-suite 0.0.1       /home/gdbeckstein/Documents/mbed-os-bluetooth-integration-testsuite
mbed-cli            1.10.4
mbed-cloud-sdk      2.0.8
mbed-flasher        0.10.1
mbed-greentea       1.7.4
mbed-host-tests     1.5.10
mbed-ls             1.7.12
mbed-os-tools       0.0.15
milksnake           0.1.5
monotonic           1.5
numpy               1.19.1
packaging           20.4
path-and-address    2.0.1
pc-ble-driver-py    0.14.2
Pillow              7.2.0
pip                 20.2
pkg-resources       0.0.0
pluggy              0.13.1
prettytable         0.7.2
protobuf            3.5.2.post1
psutil              5.6.6
py                  1.9.0
pyasn1              0.2.3
pycparser           2.20
pycryptodome        3.9.8
pyelftools          0.25
Pygments            2.7.1
PyHamcrest          2.0.2
pyparsing           2.4.7
pyrsistent          0.16.0
pyserial            3.4
pytest              6.1.1
python-dateutil     2.8.1
python-dotenv       0.14.0
pyusb               1.0.2
PyYAML              4.2b1
requests            2.20.1
semver              2.10.2
setuptools          46.1.3
six                 1.12.0
soupsieve           2.0
toml                0.10.1
trollius            2.1.post2
Twisted             20.3.0
txdbus              1.1.2
urllib3             1.24.2
Werkzeug            1.0.1
wheel               0.34.2
wrapt               1.12.1
yattag              1.13.2
zipp                3.1.0
zope.interface      5.2.0
```

#### How is this defect reproduced ? 

Connect to any Mbed OS BLE device and attempt to find the **mandatory** CCCD for the service changed characteristic.

@pan- @paul-szczepanek-arm

pan-, 2020-11-19:
Thanks @AGlass0fMilk That's a good report, I was thinking this characteristic was removed (assumption that the services never change). There is three options to solve it: 
- Remove the characteristic but it is not correct if the services on the device change 
- Implement logic to store CCCD value per client and structure of the service in persistent memory, we should also compare the previous database seen by the bonded pair with the new one and notify of changes. This is a *large* chunk of work. 
- Add the CCCD and send the notification for the whole service whenever a bonded client is connected. This can be extended latter once the Database Hash characteristic is added to BLE API providing update only when the service has changed (but still for the whole server). 

We should have bandwidth in 15 days to implement option 3. In the meantime, if it causes issue, I suggest you apply option 1 locally. 

> When I attempt to pair with a more recent Android version the connection is actually terminated! I'm thinking this issue may be related. Android likely attempts to enable indications for the service [...]

What is your version of Android or iOS ? We haven't seen any for bonded devices issue since the integration of the host privacy support. 

Side note: Just like storing value of CCCDs for bonded devices, the _Service Changed_ characteristic has often be poorly implemented across stacks especially embedded one as it requires extra space on flash. For stack implementing it properly , the notification sent was often for the whole server, not just for specific parts that changed. The addition of the DB hash from BT5 is welcomed to that regard as it is much simpler to implement in constrained environment, it also avoid re-discovery of services for a same class of devices.

AGlass0fMilk, 2020-11-19:
@pan- 

I have tested with Android 7 and Android 10. I personally do not have any issues pairing (_bonding_ however remains an issue for me on both iOS and Android) with a Galaxy S10 Lite (Android 10) or an LG Rebel 3 (Android 7).

My coworker is using a Samsung Galaxy S10 with the following specs:

[image]

In our application I have a authorization requirements for some characteristics that cause the Light Blue app to immediately attempt pairing. I'm not sure if this would cause issues.

Regarding the bonding issues:

I am unable to reconnect without performing pairing again on both the Galaxy S10 Lite and an iPhone 12 Pro running iOS 14.1

In the case of the iPhone, I have to actually unpair before I can reconnect to a device that has been power cycled. Even when the credentials should have been stored in the file system that I have initialized.

Do you have an example that you use to test pairing/bonding functionality with mobile operating systems? The current `mbed-os-example-ble` examples with the security manager and privacy aren't really aimed at this testing.

pan-, 2020-11-20:
@AGlass0fMilk Thanks for the detailed feedback, maybe this should be its own issue! 

We didn't found any issue reconnecting and re-encrypting the connection on iOS 13, 14 (not iPhone 12 pro though!), a Galaxy S20 and various Android phones from version 7 to 10. 

The BLE [security and privacy](https://github.com/ARMmbed/mbed-os-example-ble/tree/development/BLE_SecurityAndPrivacy) example should work with a phone, the device stays in the same role after bonding. It disconnects from the mobile after 3 seconds and restart advertising 3 seconds after the disconnection. 

> In the case of the iPhone, I have to actually unpair before I can reconnect to a device that has been power cycled. Even when the credentials should have been stored in the file system that I have initialized.

Is it possible to reconnect if the device has not been power cycled ? If the answer is yes, it can be a persistence issue.

AGlass0fMilk, 2020-11-20:
>     * Remove the characteristic but it is not correct if the services on the device change

I think the service should be removed by default.
 
>     * Implement logic to store CCCD value per client and structure of the service in persistent memory, we should also compare the previous database seen by the bonded pair with the new one and notify of changes. This is a _large_ chunk of work.

I think this can be implemented as a utility library _on top of_ the standard BLE API. A good candidate for something we can implement in [ARMmbed/mbed-os-experimental-ble-services](https://github.com/ARMmbed/mbed-os-experimental-ble-services).

The main consideration in this case is reserving a segment of ATT handles for the Service Changed Characteristic and CCCD. The BT spec says it must remain the same once a peer has been bonded with. We will have to discuss the best way to go about this. Is it okay for there to be gaps in the ATT handles (if the Service Changed Characteristic is not used)?
 
>     * Add the CCCD and send the notification for the whole service whenever a bonded client is connected. This can be extended latter once the Database Hash characteristic is added to BLE API providing update only when the service has changed (but still for the whole server).

Interesting, I did not notice the introduction of the Database Hash characteristic in BLE 5. I will have to review.
 
> We should have bandwidth in 15 days to implement option 3. 

I would like more control from the application over this. Some simple uses cases may not need this extra code. With clear documentation, we can make users aware of the implications of including this service and how it should be used. If we'll be refactoring how the Service Changed stuff works we should make sure to design it for maximum flexibility.

>In the meantime, if it causes issue, I suggest you apply option 1 locally.

I will try this.

> Side note: Just like storing value of CCCDs for bonded devices, the _Service Changed_ characteristic has often be poorly implemented across stacks especially embedded one as it requires extra space [...]

Yes, I can see this being the case... it's quite an esoteric part of BLE that most "users" of BLE (and even advanced BLE developers)  would not think too much about. Cache invalidation and rediscovery should be handled without so much manual work!

The reason I even noticed this issue was that I introduced a new characteristic on a device that I was testing a Linux BLE app with. Linux failed to notice the new characteristic because it had cached the old service structure! So I started looking at using the Service Changed characteristic and then noticed there was no CCCD!

> We didn't found any issue reconnecting and re-encrypting the connection on iOS 13, 14 (not iPhone 12 pro though!), a Galaxy S20 and various Android phones from version 7 to 10.

Does this include reconnecting after a power cycle using persistent storage for the bonding credentials?
 
> The BLE [security and privacy](https://github.com/ARMmbed/mbed-os-example-ble/tree/development/BLE_SecurityAndPrivacy) example should work with a phone, the device stays in the same role after [...]

I will look at the development branch of this repo and try out some of the updated examples and let you know.

> Is it possible to reconnect if the device has not been power cycled ? If the answer is yes, it can be a persistence issue.

Yes, it is possible to reconnect if the device has not been power cycled.

It would be nice to have an official example that tests _persistent_ bonding with various operating systems (Linux/Windows/Mac/iOS/Android) as these are common peers of embedded BLE devices. The nRF52840_DK (and probably other BLE targets) include on-board memory so the persistence should be enabled by default.

I would be happy to help test and work on such an example. Interoperability with mobile/desktop OS's over BLE has been a major pain point for me with Mbed's BLE implementation since I started using it in 2017 :confused:

AGlass0fMilk, 2020-11-20:
It seems there is a related issue already reported, #11809

pan-, 2020-11-20:
Thanks for the feedback @AGlass0fMilk . 

Today it would be possible to move the Generic Attribute Profile outside of Mbed OS there's no dependencies to internal component. However as soon as the DB hash is implemented I think it would be easier to have it built inside the BLE API  because we want to store the hash in the DB and store for each bond the last hash used while connected. 
Having the service builtin also allow the API to send an indication when the server is modified; otherwise we would need an event for when the server changes or the application should manually trigger this.

At the same time, having this service as external reduces the code of the main API. I need to think about it.    

> I would like more control from the application over this. Some simple uses cases may not need this extra code

Why would it be important for the application to control notification when the server change ? 

> Yes, it is possible to reconnect if the device has not been power cycled.

@paul-szczepanek-arm You might be interested by this 😁

AGlass0fMilk, 2020-11-20:
> Thanks for the feedback @AGlass0fMilk .
>
> Today it would be possible to move the Generic Attribute Profile outside of Mbed OS there's no dependencies to internal component. However as soon as the DB [...]

I think it makes sense to integrate into the main API as long as it is fully functional and can be enabled through configuration. That is to say: the inclusion of the services changed characteristic (and the generic attribute service as a whole) should be optional. This would let us have both the memory savings when don’t need it and the usefulness of this feature when we do. The documentation for this feature (and the annoying, somewhat transparent caveats due to database caching) should be discussed in adequate detail on a dedicated section of the BLE docs. Configuration of the stack in general probably needs more documentation and examples.

> > I would like more control from the application over this. Some simple uses cases may not need this extra code
> 
> Why would it be important for the application to control notification when the server change ?

It won’t be if the internal implementation works correctly 😁

I agree with you that integrating it into the main API would be simpler, since it could take advantage of the existing database API. The static handles are also something that would make it more difficult to implement outside the main API.

> > Yes, it is possible to reconnect if the device has not been power cycled.
> 
> @paul-szczepanek-arm You might be interested by this 😁

I will try out the new examples and then make a repo that your team can use to try and repeat my testing with android/iOS and the nrf52840dk

aguaviva, 2023-05-29:
This means BLE notifications never really worked. Too bad this got dismissed as low priority.