ARMmbed/mbed-os #13938: USBSerial can't work on bluepill

wubaojun1982, 2020-11-20:
### Description of defect

On bluepill (stm32f103c8t6) board, the USBSerial can't work now. The serial port can be identified. Maybe there are some usb configration errors.  My bluepill board works well with mbedos2 (https://os.mbed.com/users/hudakz/code/STM32F103C8T6_USBSerial).
#### Target(s) affected by this defect ?
Bluepill (STM32F103C8T6) boards

#### Toolchain(s) (name and version) displaying this defect ?
None.

#### What version of Mbed-os are you using (tag or sha) ?
Mbedos 6.5 (  f227856 )

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
Mbed studio 1.3.1
#### How is this defect reproduced ? 
1. The bluepill(stm32f1c8t6) ,change the PA_12 pull resistor (R10) from 10kohm to 1.5kohm.
2. Cloned project from https://os.mbed.com/users/hudakz/code/mbed-os-bluepill/
3. Config the custom_targets.json and mbed_app.json like #13838.
4. USBSerial still don't work well.

0xc0170, 2020-11-23:
cc @ARMmbed/team-st-mcd Please review

wubaojun1982, 2020-11-24:
Now I found the connection is not stable. Sometimes a plug and pull will connect successfully.

jeromecoutant, 2020-12-04:
@Allmoz @masoudr @JojoS62 
Could you check ? What do you think ?
Thx

JojoS62, 2020-12-04:
Yes, I will check tomorrow.

masoudr, 2020-12-04:
Currently, I don't have a Bluepill but for the F103RBT6 I faced a similar issue but after adding a 1.5K pull-up resistor to the USBDP line it fixed somehow. I also need to say that without the resistor I couldn't get the device connected, I mean the Windows didn't recognize it as a COM port. I don't think the type of serial tool you use is affected by this issue.
BTW here is my configuration:
```json
{
    "requires": ["bare-metal", "events", "drivers-usb"],
    "target_overrides": {
        "*": {
            "target.c_lib": "small",
            "target.printf_lib": "minimal-printf",
            "platform.minimal-printf-enable-floating-point": false,
            "target.device_has_add": ["USBDEVICE"]
        }
    }
}
```
In `USBSerial` class there is a member for blocking called `connect_blocking`, and I think you need to disable it to make sure that your code runs when you haven't connected the USB port.
```
USBSerial serial(false);
```

Allmoz, 2020-12-05:
> @Allmoz @masoudr @JojoS62
> Could you check ? What do you think ?
> Thx

I can confirm that indeed the DTR is needed to signal the device that you are ready to receive data. I can't say is a bug.

Here is a little C# .Net example to do so:

`using System;
using System.IO.Ports;
using System.Threading;

public class PortChat
{
    static SerialPort _serialPort;

    public static void Main()
    {
        Thread readThread = new Thread(Read);

        _serialPort = new SerialPort
        {
            PortName = "COM44",
            BaudRate = 115200
        };

        _serialPort.Open();

        readThread.Start();

        _serialPort.DtrEnable = true;

        _serialPort.RtsEnable = true;

        while (true)
            _serialPort.WriteLine(Console.ReadLine());
    }

    public static void Read()
    {
        while (true)
        {
            if (_serialPort.BytesToRead > 0)
            {
                try
                {
                    Console.WriteLine(_serialPort.ReadExisting());
                }
                catch (TimeoutException) { }
            }
        }
    }
}
`

Allmoz, 2020-12-05:
What I can see is that the Mbed USBSerial is not rising DSR signal.

So this does not work:
```
        while (true)
            if (_serialPort.DsrHolding)
                _serialPort.WriteLine(Console.ReadLine());
```
I found this old post too -> https://os.mbed.com/questions/75218/DSR-no-longer-raised/

jeromecoutant, 2021-02-23:
Hi
I am sorry, but I don't see what is expected from ST ? from ARM ? from ...?

JojoS62, 2021-02-23:
I see also that there is no DSR handling in USBCDC.cpp (target independent). I have checked also old sources from 2015, the same. But I'm not deep inside USB stuff to fix it quickly.

Allmoz, 2021-02-23:
> Hi
> I am sorry, but I don't see what is expected from ST ? from ARM ? from ...?

As I understand it, the OS sees the emulated serial as an RS232. RS232 has a series of signals for receiving and sending data. At this moment the implementation is inconsistent it asks for DTR (Data Terminal Ready) to send data,  but it never sets DSR (Data Set Ready)  to say that is ready to send data.

It could be USBCDC related or driver related, I'm not really sure.

MarceloSalazar, 2021-02-24:
Thanks for raising this issue. This is something for us to look at in the USB stack.
In the meantime, the workaround is to use an alternative serial monitor tool such as putty.

ithinuel, 2021-03-01:
Hello,
From what I understand from the USBCDC specs and our implementation in mbed, it seems to me that what is missing to let the PC know that the device has some data to send is the emission of the `SERIAL_STATE` interrupt (see chapter 6.5.4 from [CDC Subclass for PSTN Devices](https://www.usb.org/document-library/class-definitions-communication-devices-12)).

I don't think the CDCSerial class should send the DSR signal on its own. This should be done via a new function exposed by the CDCSerial class to trigger that event.

Also, RTS/CTS are used to throttle the communication from the DTE (aka PC) to the DCE (microcontroller). This is useful as the micro controller is much slower than the PC. DTR/DSR are used to control the flow of data in the other way (from micro controller to PC). This is less useful as the PC is expected to be much faster than the device and not require any pause in the processing of the data.

```
RTS: PC has data pending for device
CTS: Device is waiting for data
DTR: PC is waiting for data
DSR: Device has data pending for device
```

Now:
```c#
while (true)
    if (_serialPort.DsrHolding)
        _serialPort.WriteLine(Console.ReadLine());
```
isn't correct.
This makes the PC wait for the device to say "Device has data pending for PC" to itself send data.
But because the PC has already set DTR ("PC is waiting for data") the device does not have to emit a DSR signal before sending the data.

Instead of emitting a DSR (and buffering the data) when DTR is not set, the current implementation aborts the `AsyncWrite` operation and `USBCDC::send` returns `false` to let the sender that something went wrong. 
So, this is the intended behaviour.

I hope this clarifies the situation.

MarceloSalazar, 2021-03-12:
Considered resolved. Please comment if you have more questions.

closed 2021-03-12 by MarceloSalazar