ARMmbed/mbed-os #13943: Error: Identifier "I2CSlave" is undefined in MAX32630FTHR

UchiyKei1320, 2020-11-23:
Hi, I'm about to connect raspberry pi 4 as master and MAX32630FTHR as slave with I2C.
However, in MAX32630FTHR, it was outputted the message ; "Error: Identifier "I2CSlave" is undefined in "main.cpp", Line: 14, Col: 2".
I wrote definitely "#include "mbed.h"".
The photograph below is the error message.
[image]
Do you know anything about it?

0xc0170, 2020-11-23:
@UchiyKei1320 is i2c slave supported by your target? I would say it is not.

Please keep the issue template (we process the report via scripts).

0xc0170, 2020-11-23:
As this is not technical issue but rather a question, I'll close this. Please use forums.mbed.com

closed 2020-11-23 by 0xc0170