ARMmbed/mbed-os #13956: u-blox EVK-ODIN-W2

Hphipps, 2020-11-24:
### Description of defect
u-blox EVK-ODIN-W2 not supported in MBED OS 6.

#### Target(s) affected by this defect ?
u-blox EVK-ODIN-W2

#### Toolchain(s) (name and version) displaying this defect ?
n/a

#### What version of Mbed-os are you using (tag or sha) ?
mbed-os-6.5.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
n/a

#### How is this defect reproduced ? 
n/a

0xc0170, 2020-12-03:
Hi @Hphipps . I assume this is a question about why ODIN was removed from Mbed OS 6?

cc @ARMmbed/team-ublox 

This is a technical issues tracker, I'll close this.

closed 2020-12-03 by 0xc0170

Hphipps, 2020-12-03:
Would it be possible to port mbed 6 to ODIN?

0xc0170, 2020-12-03:
You can port it locally and use it as custom target.  

@ARMmbed/team-ublox shall answer the question