ARMmbed/mbed-os #13967: STM32 I2C read - every second fails (OS6)

pilotak, 2020-11-26:
### Description of defect
When trying to read a byte from I2C every first call `read()` is done. The second call is not actually done it only returns ACK: 1 even if i put a 1s delay between those two calls. This is a problem in OS6, in OS5.15 this works as intended.

#### Target(s) affected by this defect ?
STM32F412

#### Toolchain(s) (name and version) displaying this defect ?
GCC_ARM

#### What version of Mbed-os are you using (tag or sha) ?

483db06a5f78bdb2b1e1a16a82bcd1f992952306

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli

#### How is this defect reproduced ?
An example read from I2C eeprom
```cpp
#include "mbed.h"

I2C i2c(PB_7, PB_6);

int main() {
    char buf[1];

    for (size_t i = 0; i < 2; i++) {
        int ack = i2c.read(0x50 << 1, buf, 1);
        printf("ack=%i\n", ack);
        // ThisThread::sleep_for(1s);
    }

    return 0;
}
```
add debug to I2C read()
https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/i2c_api.c#L914-L920

```diff
if ((timeout == 0) || (obj_s->event != I2C_EVENT_TRANSFER_COMPLETE)) {
    DEBUG_PRINTF(" TIMEOUT or error in i2c_read\r\n");
    /* re-init IP to try and get back in a working state */
    i2c_init_internal(obj, NULL);
+   printf("timeout\n");
} else {
    count = length;
}
```

## Results:
This is OS6
```
ack=0
timeout
ack=1
```
[image]

This is OS5.15.6 (b114a9c878519d6489ac3426697196bbea34c8ea)
```
ack=0
ack=0
```
[image]

jeromecoutant, 2020-11-27:
> This is a problem in OS6, in OS5.15 this works as intended

Is it possible to check before and after #12626 ?
Thx

pilotak, 2020-11-30:
@jeromecoutant I can confirm it works fine before (e8208c062b44075d3e5565e4510b21aa05932caa) and if fails after CUBE update (6678efe571d9d3c28c9853ac2ea875d5c01664cc)

pilotak, 2021-01-05:
@jeromecoutant any update?

jeppefrandsen, 2021-01-22:
We are seeing the same issue. Probably need to go back in versions for now.

jeppefrandsen, 2021-01-23:
It works for me with 1.23.0 but not 1.24.2. Maybe you can check also @pilotak?

ikmdani, 2021-03-01:
**Reproducing the problem:**
From the “description of defect” I understand that you want to read EEPROM using i2c bus operations. Here I try to reproduce the problem using the code as below, using an accelerometer instead of EEPROM as i2c slave device connected to NUCLEO-F29ZI target platform with mbed-OS 6.8.0.

[image]

Below is the bus activity for the above code, it is exactly as mentioned in the problem description.
[image]

**Analysis:**
After looking at the code and executing the code, I can confidently say that the code mentioned is not the correct way of reading data from i2c slave devices.

The correct way to perform i2c read is to first call i2c.write() with a mention of register offset (0x2C in my example) to be read and followed by i2c.read() with last parameter mentioning number of bytes to be read. The below code reads 2 bytes of data starting from register offset 0x2C. Here i2c.read() function is called only once and after successfully executing this line, the buffer cmd contains the bytes read.

[image]

Below is the bus activity for the above logic, data read is 0x0A (at 0x2C) and 0x00 (at 0x2D). 

[image]

If multiple i2c read operations needs to be performed then the followed shall be the way. Here I am trying to read 2 bytes of data starting from register offset 0x2C, three times.

[image]
Below is the bus activity for the above logic. I could not capture the complete timing diagram here.

[image]
**Conclusion:**
I didn’t see any problem as such here, it is working as expected. So this ticket can be closed if this justification is acceptable.

pilotak, 2021-03-01:
There are devices where you don't need to specify offset you want to read from - it is increased in slave device automatically. So i don't think you are right. Another example: you have a limited buffer to read to (which could be the ie. reading 4k data from slave with 64 bytes allocated buffer, you just process it gradually - you don't need to waste 4k of RAM), there is no need to set the offset again.

The problem here is that this worked fine before upgrading the STM32Cube, not if the code is correct - it was only for reproducing the issue!

0xc0170, 2021-05-17:
The fix was referenced above, please review

pilotak, 2021-05-19:
I can confirm that the fix solves this issue

rotu, 2021-05-20:
Does this issue/fix affect the async `transfer` API?

jeromecoutant, 2021-05-20:
> Does this issue/fix affect the async `transfer` API?

@ikmdani maybe you have some comment ? Thx

ikmdani, 2021-05-20:
@rotu 

> Does this issue/fix affect the async `transfer` API?

This commit is for STM32's F2, F4 and L1 series only. it won't affect negatively if you have not faced any issue during i2c read before this PR.

rotu, 2021-05-20:
I’m currently having lots of trouble getting I2C to work on an F446. The first rx-only transfer succeeds and all subsequent ones fail. I can’t tell if it’s user error, hardware error, or library error.

So regardless of the fix does this *bug* affect the async API?

ikmdani, 2021-05-21:
> The first rx-only transfer succeeds and all subsequent ones fail.

@rotu, Would you mind sharing your code.

rotu, 2021-05-21:
My code is basically:

```
EventFlags f;
int res = i2c.transfer(addr, nullptr, 0, buf, buflen, [&f](int){f.set(1);});
int wait_res = f.wait_all_for(1,100ms);
```

I’ve also tried adding `i2c.abort_transfer()` and `i2c.start()` before this code, to no avail.

My problem could very well be the device I’m communicating with - it’s a BNO080, which uses i2c in a strange way (not supporting repeated start conditions, for instance).

closed 2021-05-24 by adbridge

rotu, 2021-05-24:
So no word on whether this affects the asynch transfer API? Nor whether the fix to this affected that API?

ikmdani, 2021-05-24:
@rotu i have used the below code to reproduce your case-scenario. With ARMmbed/mbed-os#14668, it works well. I am able to read the ADXL sensor. `i2c.transfer` is called 6 times in this example with each time it reads 10 bytes.

```
#include "ThisThread.h"
#include "mbed.h"

I2C i2c(PB_9, PB_8);

const int addr7bit = 0x53;      // 7 bit I2C address
const int addr8bit = 0x53 << 1; // 8bit I2C address

static bool i2c_debug_busy = false;
void callback_debug(int a)
{
        i2c_debug_busy = false;
}

int main() {
    int ack = 0;
    char cmd[10];
    char i2c_tx[10];
    char i2c_rx[10];
    size_t j = 0;
    int bytes_to_write = 0;
    int bytes_to_read = 10;

        printf("started...\r\n");

        for(size_t i=0;i<6;i++)
        {
                for(j=0;j<bytes_to_read; j++)
                        i2c_rx[j] = 0;

                i2c_debug_busy = true;

                ack = i2c.transfer(addr8bit, &i2c_tx[0], bytes_to_write, &i2c_rx[0], bytes_to_read, callback_debug);
                ThisThread::sleep_for(1);

                printf("\r\nR ack= %d\t", ack);

                for(j=0;j<bytes_to_read; j++)
                        printf("0x%x ", i2c_rx[j]);
        }

        printf("\r\n=============END================\r\n");
}
```

PR #14668 (2021-05-24): Driver: I2C: STM32F2/STM32F4/STM32L1: Fix alternate i2c read.

This commit fixes the i2c driver issue reported in https://github.com/ARMmbed/mbed-os/issues/13967 on STM32F2xx, STM32F4xx & STM32L1xx series.
fix #13967 

### Summary of changes 
As issue describes, every alternate i2c read was failing. On the bus side there was no transaction of alternate i2c read, the state machine (`PreviousState`) prevented to trigger START condition of second transaction as the first transaction did not end it cleanly, the `PreviousState` was still set as `I2C_STATE_MASTER_BUSY_RX` although the first transact [... cut]