ARMmbed/mbed-os #13980: Calling non-void function that has no defined return crashes app when compiled by newer GCC Arm

zhiyong80, 2020-11-28:
### Description of defect

When a non-void function that has no defined return is called by another function, the whole app will crash. This seems to happen with newer version of GCC Arm compiler such as V9, but not older version such as V6 2017-q1-update. I am not entirely sure if this is a newly introduced feature of GCC Arm compiler itself or this is a bug. It makes sense to make sure a return is defined for a non-void function, displaying a warning message is nice. But crashing the whole code for this seems a bit excessive and dramatic.

#### Target(s) affected by this defect ?

NUCLEO_F207ZF, NUCLEO_F429ZI

#### Toolchain(s) (name and version) displaying this defect ?
GCC Arm V9 2019-q4-major

#### What version of Mbed-os are you using (tag or sha) ?

Mbed-os-5.15.3

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

mbed-cli 1.8.3

#### How is this defect reproduced ? 

1) Compile the following code for NUCLEO_F429ZI with GCC_ARM V9

```
#include "mbed.h"

uint8_t sendCmdResponse(uint8_t response[], uint8_t responseLen);

int main()
{
    uint8_t response[] = "This is a test CMD response.\r\n";
    while (1) {
        printf("This is a test CMD response.\r\n");
        ThisThread::sleep_for(1000);
        sendCmdResponse(response, sizeof(response) - 1);
        //ThisThread::sleep_for(1000); // Different crash info if delay is placed here
    }
}

uint8_t sendCmdResponse(uint8_t response[], uint8_t responseLen)
{
    // Place holder
}
```
2) Connect a terminal to NUCLEO_F429ZI board.

3) Copy built bin file to NUCLEO_F429ZI board

4) There should be some sort of crash information in terminal depending on where the delay line is placed. If there is no delay between printf() and sendCmdResponse(), I get the following information.
```
This is a test CMD response.
assertion "" failed: file "", line 0
```
If there is a delay between those two lines, I get the following information.
```
This is a test CMD response.
assertion "
++ MbedOS Fault Handler ++

FaultType: HardFault

Context:
R0   : FFFFFFF8
R1   : 00890000
R2   : FFFFFFFF
R3   : 00000000
R4   : 00000000
R5   : 0000002B
R6   : 00890000
R7   : 00000000
R8   : 2000741C
R9   : 2000082C
R10  : 080109E7
R11  : 20000470
R12  : FFFFFFFF
SP   : 200073C8
LR   : 0800B031
PC   : 08008824
xPSR : 61010000
PSP  : 20007360
MSP  : 2002FFC0
CPUID: 410FC241
HFSR : 40000000
MMFSR: 00000000
BFSR : 00000082
UFSR : 00000000
DFSR : 00000008
AFSR : 00000000
BFAR : 00890000
Mode : Thread
Priv : Privileged
Stack: PSP

-- MbedOS Fault Handler --

++ MbedOS Error Info ++
Error Status: 0x80FF013D Code: 317 Module: 255
Error Message: Fault exception
Location: 0x8008824
Error Value: 0x200001B0
Current Thread: main Id: 0x20004E38 Entry: 0x800294B StackSize: 0x2000 StackMem: 0x20005528 SP: 0x200073C8 
For more info, visit: https://mbed.com/s/error?error=0x80FF013D&tgt=NUCLEO_F429ZI
-- MbedOS Error Info --

= System will be rebooted due to a fatal error =
= Reboot count(=1) reached maximum, system will halt after rebooting =.
```
6) Change GCC Arm compiler version to V6 2017-q1-update. Then delete build folder, re-compile the same code for NUCLEO_F429ZI board, and copy resulted bin file to the board.

7) The output in terminal I get is repeated output of lines test string.
```
This is a test CMD response.
This is a test CMD response.
This is a test CMD response.
This is a test CMD response.
This is a test CMD response.
This is a test CMD response.
This is a test CMD response.
This is a test CMD response.
This is a test CMD response.
This is a test CMD response.
........
```

zhiyong80, 2020-11-28:
@ciarmcom Sorry I just grabbed the result of git describe and forgot about hash. It is updated now.

zhiyong80, 2020-11-28:
Just tried the test code on K64F with mbed-os-6.5.0-93-g33a7e66a07, app crashes after the first run of cycle, with the following crash info:
```
This is a test CMD response.

++ MbedOS Error Info ++
Error Status: 0x80FF0144 Code: 324 Module: 255
Error Message: Assertion failed: status == osOK
Location: 0x34D9
File: .\mbed-os\rtos\source\ThisThread.cpp+225
Error Value: 0x0
Current Thread: main Id: 0x20001270 Entry: 0x2595 StackSize: 0x2000 StackMem: 0x20001E48 SP: 0x20003DD8 
For more info, visit: https://mbed.com/s/error?error=0x80FF0144&tgt=K64F
-- MbedOS Error Info --

= System will be rebooted due to a fatal error =
= Reboot count(=2) reached maximum, system will halt after rebooting =
```

zhiyong80, 2020-11-28:
This also happens to ARCH_MAX + mbed-os-6.5.0-93-g33a7e66a07. This is probably not a release/board specific issue.
```
++ MbedOS Error Info ++
Error Status: 0x80FF0144 Code: 324 Module: 255
Error Message: Assertion failed: status == osOK
Location: 0x8003C89
File: .\mbed-os\rtos\source\ThisThread.cpp+225
Error Value: 0x0
Current Thread: main Id: 0x2000136C Entry: 0x80023CD StackSize: 0x2000 StackMem: 0x200053D8 SP: 0x2000736C 
For more info, visit: https://mbed.com/s/error?error=0x80FF0144&tgt=ARCH_MAX
-- MbedOS Error Info --
```

chrisc11, 2020-11-30:
For C++, a missing return statement will result in undefined behavior. (For what it's worth there's been some interesting discussion about it over the years in [GCCs bugzilla](https://gcc.gnu.org/bugzilla/show_bug.cgi?id=43943).)

It does look like for more recent versions of GCC the behavior is now more truly undefined. If I compile a single file with your example from above I see g++ wind's up removing the `while(1) { }` loop!

```
// @file test.c
uint8_t sendCmdResponse(uint8_t response[], uint8_t responseLen) {
    // Place holder
}

int main(void) {
  while (1) {
    uint8_t response[] = "This is a test CMD response.\r\n";
    sendCmdResponse(response, sizeof(response) - 1);
  }
}
```

```
# Using arm-none-eabi-g++ (GNU Tools for ARM Embedded Processors) 5.4.1 20160919
$ arm-none-eabi-g++  -Os -c test.c && arm-none-eabi-objdump -C -d test.o
test.o:     file format elf32-littlearm
Disassembly of section .text:

00000000 <sendCmdResponse(unsigned char*, unsigned char)>:
   0:	e12fff1e 	bx	lr

Disassembly of section .text.startup:
00000000 <main>:
   0:	eafffffe 	b	0 <main>
```

```
# Using arm-none-eabi-g++ (GNU Arm Embedded Toolchain 9-2020-q2-update) 9.3.1 20200408
$ arm-none-eabi-g++  -Os -c test.c && arm-none-eabi-objdump -d -C test.o
test.c: In function 'uint8_t sendCmdResponse(uint8_t*, uint8_t)':
test.c:5:1: warning: no return statement in function returning non-void [-Wreturn-type]
    5 | }

test.o:     file format elf32-littlearm
# empty ... everything got removed, even unrelated code!
```

> I am not entirely sure if this is a newly introduced feature of GCC Arm compiler itself or this is a bug. It makes sense to make sure a return is defined for a non-void function, displaying a warning message is nice. But crashing the whole code for this seems a bit excessive and dramatic.

Definitely feels excessive but because you are falling into an undefined area of the specification, I believe the behavior is more or less "expected" (sadly). The best way to resolve the issue would be to fix any `-Wreturn-type` warning's triggered by the compiler / promote them to compilation errors by adding `-Werror=return-type` to GCC_ARM->cxx for the mbed profile you are using to compile.

zhiyong80, 2020-11-30:
Hi @chrisc11 , thanks for the information. Being self-taught and all over the place, I didn't know much about the background around this. Good thing I didn't ignorantly declare this as a bug outright. LOL

chrisc11, 2020-11-30:
No problem @zhiyong80, definitely a confusing issue! I was looking through the mbed issues list for something else and your title piqued my curiosity so I took a quick look :)

0xc0170, 2020-11-30:
Thanks @chrisc11 for helping.

I believe this is answered and I'll close it.

closed 2020-11-30 by 0xc0170

zhiyong80, 2020-11-30:
It is probably better to promote the warning message to error as suggested by Chris in newer version of mbed-cli. The way it is will trip someone else later.

0xc0170, 2020-11-30:
There was already reported earlier to add some additional flags. Once we find time to update, this return type could be added as well.