ARMmbed/mbed-os #13982: NRF52 ADC Difference Mode

liningpan, 2020-11-30:
### Description of defect

NRF52840 has SAADC that support differential input. The result is a 16-bit integer in 2's complement form. However the analog read function will check and ignore negative values. 

https://github.com/ARMmbed/mbed-os/blob/33a7e66a0794c91fddc19f9217a81910d5f4a23f/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_NRF52/analogin_api.c#L130-L150

This makes the differential input functionality provided by the microcontroller useless. 

#### Target(s) affected by this defect ?
NRF52840 Arduino Nano BLE 33

#### Toolchain(s) (name and version) displaying this defect ?
Arduino mbed core 1.3.0
https://github.com/arduino/ArduinoCore-mbed/releases/tag/1.3.0

#### What version of Mbed-os are you using (tag or sha) ?

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

#### How is this defect reproduced ?

AGlass0fMilk, 2020-11-30:
@liningpan The Mbed OS AnalogIn API does not currently support differential input. 

You are better off using the ADC HAL API directly and avoid using `mbed::AnalogIn`. Otherwise, you will have to locally modify the analogin_hal.c implementation to use differential inputs and be able to handle negative 16-bit values.

liningpan, 2020-11-30:
@AGlass0fMilk I see. Thank you! I will close this issue.

closed 2020-11-30 by liningpan