ARMmbed/mbed-os #13983: CMake GCC ARM toolchain script specifies the wrong target variable for linker script and map file paths

sa-nloika, 2020-11-30:
### Description of defect

The CMake toolchain function `mbed_set_toolchain_options()` in `tools/cmake/toolchains/GCC_ARM.cmake` construct the linker script and map file paths using the variable `${APP_TARGET}` instead of the function argument `${target}`. This can cause improper files outputs if `APP_TARGET` is not defined.

https://github.com/ARMmbed/mbed-os/blob/master/tools/cmake/toolchains/GCC_ARM.cmake#L22

#### Target(s) affected by this defect ?

all

#### Toolchain(s) (name and version) displaying this defect ?

GCC ARM

#### What version of Mbed-os are you using (tag or sha) ?

33a7e66a0794c91fddc19f9217a81910d5f4a23f

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

mbed-tools, 4ed041a416e6b8a08eb22ffdfdd2bb111c2f2c84

#### How is this defect reproduced ? 

NA

ladislas, 2020-11-30:
I can confirm the issue. I tried changing `${APP_TARGET}` to `${target}` but it doesn't seem to work.

Have you had any success?

0xc0170, 2020-11-30:
Thanks for the report, we will triage this.

> I can confirm the issue. I tried changing ${APP_TARGET} to ${target} but it doesn't seem to work.

What does this mean, using the argument does not work - the path is wrong, it fails or ?

ladislas, 2020-11-30:
@0xc0170 with the following patch on `toolchains/GCC_ARM.cmake`

```diff
diff --git a/tools/cmake/toolchains/GCC_ARM.cmake b/tools/cmake/toolchains/GCC_ARM.cmake
index 9b9fa2062f..63a5eef3e7 100644
--- a/tools/cmake/toolchains/GCC_ARM.cmake
+++ b/tools/cmake/toolchains/GCC_ARM.cmake
@@ -19,8 +19,8 @@ function(mbed_set_toolchain_options target)
             "-lnosys"
         "-Wl,--end-group"
         "-specs=nosys.specs"
-        "-T" "${CMAKE_BINARY_DIR}/${APP_TARGET}.link_script.ld"
-        "-Wl,-Map=${CMAKE_BINARY_DIR}/${APP_TARGET}.map"
+        "-T" "${CMAKE_BINARY_DIR}/${target}.link_script.ld"
+        "-Wl,-Map=${CMAKE_BINARY_DIR}/${target}.map"
         "-Wl,--cref"
     )
```

and the following on `CMakeLists.txt`

```diff
diff --git a/CMakeLists.txt b/CMakeLists.txt
index 917b959..f127905 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -5,31 +5,31 @@ cmake_minimum_required(VERSION 3.19.0 FATAL_ERROR)

 set(MBED_PATH ${CMAKE_CURRENT_SOURCE_DIR}/mbed-os CACHE INTERNAL "")
 set(MBED_CONFIG_PATH ${CMAKE_CURRENT_SOURCE_DIR}/.mbedbuild CACHE INTERNAL "")
-set(APP_TARGET mbed-os-example-blinky)
+# set(APP_TARGET mbed-os-example-blinky)

 include(${MBED_PATH}/tools/cmake/app.cmake)

 add_subdirectory(${MBED_PATH})

-add_executable(${APP_TARGET})
+add_executable(mbed-os-example-blinky)

-mbed_configure_app_target(${APP_TARGET})
+mbed_configure_app_target(mbed-os-example-blinky)

-mbed_set_mbed_target_linker_script(${APP_TARGET})
+mbed_set_mbed_target_linker_script(mbed-os-example-blinky)

-project(${APP_TARGET})
+project(mbed-os-example-blinky)

-target_sources(${APP_TARGET}
+target_sources(mbed-os-example-blinky
     PRIVATE
         main.cpp
 )

-target_link_libraries(${APP_TARGET}
+target_link_libraries(mbed-os-example-blinky
     PRIVATE
         mbed-os
 )

-mbed_set_post_build(${APP_TARGET})
+mbed_set_post_build(mbed-os-example-blinky)
```

I get:

```console
[245/245] Linking CXX executable mbed-os-example-blinky
FAILED: mbed-os-example-blinky mbed-os-example-blinky.link_script.ld
cd /Users/ladislas/dev/tmp/mbed-os-example-blinky && arm-none-eabi-cpp @/Users/ladislas/dev/tmp/mbed-os-example-blinky/cmake_build/compile_time_defs.txt -x assembler-with-cpp -E -Wp,-P /Users/ladislas/dev/tmp/mbed-os-example-blinky/mbed-os/targets/TARGET_STM/TARGET_STM32F7/TARGET_STM32F769xI/TOOLCHAIN_GCC_ARM/STM32F769xI.ld -o /Users/ladislas/dev/tmp/mbed-os-example-blinky/cmake_build/mbed-os-example-blinky.link_script.ld && cd /Users/ladislas/dev/tmp/mbed-os-example-blinky/cmake_build && /usr/local/bin/arm-none-eabi-g++  -mthumb -mfpu=fpv5-sp-d16 -mfloat-abi=softfp -mcpu=cortex-m7 -Wall -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -fmessage-length=0 -fno-exceptions -ffunction-sections -fdata-sections -funsigned-char -MMD -fomit-frame-pointer -g3 -Wl,--start-group -lstdc++ -lsupc++ -lm -lc -lgcc -lnosys -Wl,--end-group -specs=nosys.specs -T /Users/ladislas/dev/tmp/mbed-os-example-blinky/cmake_build/mbed-core.link_script.ld -Wl,-Map=/Users/ladislas/dev/tmp/mbed-os-example-blinky/cmake_build/mbed-core.map -Wl,--cref -Wl,--gc-sections -Wl,--wrap,main -Wl,--wrap,_malloc_r -Wl,--wrap,_free_r -Wl,--wrap,_realloc_r -Wl,--wrap,_memalign_r -Wl,--wrap,_calloc_r -Wl,--wrap,exit -Wl,--wrap,atexit -Wl,-n -Wl,--wrap,printf -Wl,--wrap,sprintf -Wl,--wrap,snprintf -Wl,--wrap,vprintf -Wl,--wrap,vsprintf -Wl,--wrap,vsnprintf -Wl,--wrap,fprintf -Wl,--wrap,vfprintf @CMakeFiles/mbed-os-example-blinky.rsp -o mbed-os-example-blinky  && cd /Users/ladislas/dev/tmp/mbed-os-example-blinky/cmake_build && arm-none-eabi-objcopy -O binary /Users/ladislas/dev/tmp/mbed-os-example-blinky/cmake_build/mbed-os-example-blinky /Users/ladislas/dev/tmp/mbed-os-example-blinky/cmake_build/mbed-os-example-blinky.bin && /usr/local/Cellar/cmake/3.19.1/bin/cmake -E echo "-- built: /Users/ladislas/dev/tmp/mbed-os-example-blinky/cmake_build/mbed-os-example-blinky.bin" && arm-none-eabi-objcopy -O ihex /Users/ladislas/dev/tmp/mbed-os-example-blinky/cmake_build/mbed-os-example-blinky /Users/ladislas/dev/tmp/mbed-os-example-blinky/cmake_build/mbed-os-example-blinky.hex && /usr/local/Cellar/cmake/3.19.1/bin/cmake -E echo "-- built: /Users/ladislas/dev/tmp/mbed-os-example-blinky/cmake_build/mbed-os-example-blinky.hex" && cd /Users/ladislas/dev/tmp/mbed-os-example-blinky/cmake_build && /usr/local/Frameworks/Python.framework/Versions/3.9/bin/python3.9 /Users/ladislas/dev/tmp/mbed-os-example-blinky/mbed-os/tools/memap.py -t GCC_ARM /Users/ladislas/dev/tmp/mbed-os-example-blinky/cmake_build/mbed-os-example-blinky.map
/usr/local/Cellar/arm-gcc-bin/9-2020-q2-update/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: cannot open linker script file /Users/ladislas/dev/tmp/mbed-os-example-blinky/cmake_build/mbed-core.link_script.ld: No such file or directory
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ERROR: CMake invocation failed!
```

0xc0170, 2020-11-30:
I could reproduce, thanks. We will fix.

0xc0170, 2020-11-30:
Fix proposed https://github.com/ARMmbed/mbed-os/pull/13985

sa-nloika, 2020-11-30:
Sorry, should have made that statement more clear!

I attempted to make a project where `${APP_TARGET}` was not defined and the resulting linker script file names were `cmake_build/.link_script.ld` which caused the project to fail during linking.\

Thanks for proposing a fix so quick!

closed 2020-12-03 by 0xc0170