ARMmbed/mbed-os #13984: Can not reopen virtual serial port on some ST board with high send rate

xia-rj, 2020-11-30:
### Description of defect

I run the USBSerial example, it working well when sleep_for time is 1s, but if changed sleep_for time to 1ms, then closing and reopening the serial, can not received any message on PC(WIN10 or UBUNTU 18.04).

#### Target(s) affected by this defect ?
NUCLEO F767ZI and NUCLEO WB55. On NUCLEO WB55, after reopening the serial, sometime it can working well, sometime can't.

#### Toolchain(s) (name and version) displaying this defect ?
armcc6

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.3, mbed-os-6.5, mbed-os-5.15

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
Mbed studio 1.3.1

#### How is this defect reproduced ?

0xc0170, 2020-12-01:
cc @ARMmbed/team-st-mcd

jeromecoutant, 2021-03-25:
@xia-rj could we close this issue ?
Thx

closed 2021-05-25 by adbridge