ARMmbed/mbed-os #13989: pal_crypto.h file name conflict with Pelion client (the sequel)

ladislas, 2020-12-01:
### Description of defect

This is the exact same issue as described in https://github.com/ARMmbed/mbed-os/issues/10022.

The issue was fixed in https://github.com/ARMmbed/mbed-os/pull/10042

But then the changes were reverted and the issue brought back with https://github.com/ARMmbed/mbed-os/pull/13217 and https://github.com/ARMmbed/mbed-os/commit/3e011a9c4ac8388bdd50256f961a564b832126b0

On the cloud client end, the PR mentioned by @adamelhakham (https://github.com/ARMmbed/mbed-os/issues/10022#issuecomment-471923066) is nowhere to be found.

#### Target(s) affected by this defect ?

Custom target with BLE and Pelion

#### Toolchain(s) (name and version) displaying this defect ?

```console
arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Arm Embedded Toolchain 9-2020-q2-update) 9.3.1 20200408 (release)
```

#### What version of Mbed-os are you using (tag or sha) ?

6.5.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

```
mbed --version
1.10.4

mbed-tools --version
3.5.1.dev36
```

#### How is this defect reproduced ? 

Setup Pelion mbed-cloud-client using a target with BLE.

0xc0170, 2020-12-01:
@ARMmbed/mbed-os-connectivity Please review

ladislas, 2021-01-12:
any news on this?

0xc0170, 2021-01-12:
@ladislas I assume  this won't a problem once we switch to new tools with CMake? Meanwhile we can rename it again for the time being? 

@paul-szczepanek-arm Can you review?

paul-szczepanek-arm, 2021-01-12:
The include is part of cordio where it's included without a path - hence the conflict. I don't want to change cordio unless I really have to as it's an external project. Any chance of pelion changing its filename or using cmake to avoid the include paths mess?

ladislas, 2021-01-12:
> I assume this won't a problem once we switch to new tools with CMake? Meanwhile we can rename it again for the time being?

@0xc0170 I hope! Can't wait to have bootloader/cmake compatibility.

> Any chance of pelion changing its filename or using cmake to avoid the include paths mess?

@paul-szczepanek-arm That's what we did in the mean time, made a fork and renamed the files. But it's not ideal.

pan-, 2021-03-25:
I'm closing this as it won't be solved inside Mbed OS. 
Using the new build system should mitigate the issue until it is fixed in Mbed cloud client.

closed 2021-03-25 by pan-