ARMmbed/mbed-os #13990: Retransmission with same uplink framecounter get ignored by network servers

martinichka, 2020-12-01:
### Description of defect

Current implementation: when LoRaWAN stack does not receive confirmation on a confirmed uplink it will retry for a configurable amount of retries, without increasing the frame counter.

Network servers don't acknowledge retransmissions (with the same framecounter) due to replay attacks. So in case the uplink was received by the network server but the downlink was not received by the device it will retry in vain. 

For reference:
- https://forum.chirpstack.io/t/acks-for-retransmissions-of-confirmed-uplink/51
- https://github.com/TheThingsNetwork/ttn/issues/272

#### Target(s) affected by this defect ?
Any using LoRaWAN feature

#### Toolchain(s) (name and version) displaying this defect ?
Any

#### What version of Mbed-os are you using (tag or sha) ?

Any (did not change since the introduction up till the current master)

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
NA

#### How is this defect reproduced ? 
Periodically send LoRaWAN confirmed uplinks to a network server like ChripStack. Wait till (or force) a confirmation downlink is not receive. Then you will observe the device will retransmit the same messages for the number of configured retries, but the server will not send any confirmation downlinks anymore (as it already did on the first received uplink). On the device this will lead to `Retries exhausted for Class A device`. 

So energy is wasted without chance of success.

martinichka, 2020-12-01:
I created a PR for resolving this issue here: #13991

Eric-A-Marks, 2020-12-21:
As per the LoRaWAN spec, Page 18, https://lora-alliance.org/sites/default/files/2018-05/lorawan1_0_2-20161012_1398_1.pdf

"The  FCnt  is  not incremented  in  case  of  multiple  transmissions of  an unconfirmed frame(see  NbTrans parameter), or in the case of a confirmed frame that is not acknowledged"

If you want to increment the frame counter every uplink, set the LoRaWAN stack nb_trials to 0, and then retry the send at the application level.  Be aware that this has implications for ADR.

0xc0170, 2021-01-11:
Thanks @Eric-A-Marks . Does this mean PR 13991 is not needed?

0xc0170, 2021-01-11:
cc @ARMmbed/mbed-os-connectivity

martinichka, 2021-01-11:
This does not solve the problem. It will break the ADR. Without the fix the
MBed LoRaWAN stack (in combination with the ChripStack servier) will never
successfully receive an acknowledgement on retransmission if it failed to
receive the acknowledgement on the first transmission.

Eric-A-Marks, 2021-01-11:
@0xc0170  I would advise to not apply PR 13991 as it would not follow the specification.  It would cause to network servers to emit duplicate messages on ack loss, since each message would look unique.   The patch would also prevent LoRa Alliance device certification.

johanstokking, 2021-01-22:
I agree. I don't know why The Things Network is mentioned here as a reason to deviate from the LoRaWAN standard.

> Network servers don't acknowledge retransmissions (with the same framecounter) due to replay attacks. So in case the uplink was received by the network server but the downlink was not received by the device it will retry in vain.

This is by design. That is not specific to The Things Network or ChirpStack or any other LoRaWAN network server for that matter.

0xc0170, 2021-01-22:
> I created a PR for resolving this issue here: #13991

I closed the pull request.

Thank you all for the helpful feedback. I'll close this  issue as well as "won't fix".

closed 2021-01-22 by 0xc0170

martinichka, 2021-01-22:
> I don't know why The Things Network is mentioned here as a reason to deviate from the LoRaWAN standard.
@johanstokking I only referred to a forum discussion at TheThingsNetwork

I do understand that performing a retransmission with a new frame counter does deviate from the LoRaWAN standard. I do understand that my proposed fix (#13991) generates duplicates that cannot be de-duplicated by the network and therefore is not preferred.

But I have no solution for devices that do not receive the acknowledge. We have over 10 thousand (LoRaWAN v1.0.2) devices and this is the majority of failures. In this case if the server receives an confirmed uplink but the device does not receive the acknowledge, it retries but the retries do not get accepted by the server. It repeats for the configured number of reties, without chance of success, Is implementing a custom retransmission in the application layer the only solution for this (on LoRaWAN v1.0.2)?

@johanstokking Is error expected to be handled better with LoRaWAN 1.1? Or would it be the same for this matter?

johanstokking, 2021-01-25:
LoRaWAN applications must assume that the network layer is lossy. Confirmed messages should be retried with an incrementing frame counter, and the receiving end must be made aware of this.

Your solution was to bump FCnt for retransmissions, but those must always have the same FCnt. This is the mechanism:

>658 End-devices SHALL stop any further retransmission of an uplink confirmed frame if a
>659 corresponding downlink acknowledgment frame is received.
>
>660 End-devices SHALL also stop any further retransmission of an uplink unconfirmed frame
>661 whenever a valid downlink frame is received in a Class A receive window.
>
>662 If the Network receives more than `NbTrans` transmissions of the same uplink frame having
>663 the ADR bit set, this may indicate a replay attack or a malfunctioning end-device, and therefore
>664 the Network SHALL silently discard the extra frames.

So, after `NbTrans` retransmissions of a confirmed uplink frame, and the device _really_ needs a confirmation, try again (later) with a new frame counter.