ARMmbed/mbed-os #13997: Ambiq Micro: HciDrvIntService dependency in target HAL

0xc0170, 2020-12-02:
### Description of defect

`HciDrvIntService` is part of startup as ISR handler. This brings the dependency on BLE driver. I believe we can fix it. As the ISR handler (default one for BLEIP) should be weakly defined and implemented as the rest of ISR handlers - empty.

This is what is causing a trouble, in startup_gcc.c:

```
extern void HciDrvIntService(void); //__attribute((weak, alias("am_default_isr")));
extern void am_gpio_isr(void); //__attribute((weak, alias("am_default_isr")));
extern void am_ctimer_isr(void); //__attribute((weak, alias("am_default_isr")));
extern void am_uart_isr(void); //__attribute((weak, alias("am_default_isr")));
extern void am_uart1_isr(void); //__attribute((weak, alias("am_default_isr")));
```

As we provide NVIC_SetVector() we should use it and dynamically attach ISR handler. If BLE is enabled, it would register its handler. If not, default isr would be used.

We found this while porting target to CMake as a target does not compile without additional libraries (in this case BLE).

#### Target(s) affected by this defect ?

Ambiq_Macro

#### Toolchain(s) (name and version) displaying this defect ?

All

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.5.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

Not related

#### How is this defect reproduced ? 

Look at files mentioned above, we do not have yet PR with CMake as we hit couple of issues, this is one of them.

cc @ARMmbed/team-sparkfun

0xc0170, 2020-12-02:
@Wenn0101 Please review

0xc0170, 2020-12-02:
I can confirm if I do for GCC ARM: `extern void HciDrvIntService(void) __attribute((weak, alias("am_default_isr")));`, I don't get an undefined symbols. The next one would be to register ISR handler in `HciDrvRadioBoot(bool bColdBoot)` where we would attach proper ISR for BLE.

Wenn0101, 2020-12-02:
@0xc0170 Good catch. We removed these weak definitions because of a problem we had when we built this as a static library to include in other projects. When linking these projects we ran into a problem where the linker would link the weak symbol rather than the strong. 

Now that I see it is causing problems, I think the solution of reverting back to `extern void HciDrvIntService(void) __attribute((weak, alias("am_default_isr")));` and using `NVIC_SetVector()` sounds like a good solution.

I'll try this out and let you know if it works as expected.

0xc0170, 2020-12-10:
@Wenn0101 any update?  This is blocking CMake port PR

Wenn0101, 2020-12-15:
@0xc0170 Sorry I took a few personal days off work and did not see your latest reply until just now. My initial testing of your suggestions were looking they they would work fine for us but I hadn't yet tested to verify if they fixed the initial issue. I see there is now an open PR by @hugueskamba. I will make it my first priority today as I resume work today to review what has been done and contribute what I can. Sorry if this has caused any delay.

Wenn0101, 2020-12-17:
@0xc0170 As mentioned I have done some testing with my existing build process to verify that your suggested changes don't break anything on our end, and we have fixed the bug in our projects that use a static library of our targets that made us introduce this problem in the first place. 

What I am less in tune with is how to reproduce your original problem, to confirm that what I have done fixes this problem. I am just understanding how this should fix the problem in theory. How should I build with cmake? can I do this in the master branch or should I be doing this in a feature branch?

I have made the suggested changes [here ](https://github.com/Wenn0101/mbed-os/tree/cmake-ambiq-gcc-fix)to do some testing, but should these changes be added to [this PR](https://github.com/ARMmbed/mbed-os/pull/14042/commits/66291598cbf2ac0ebd3af864945621599c509c4d).

hugueskamba, 2020-12-17:
> @0xc0170 As mentioned I have done some testing with my existing build process to verify that your suggested changes don't break anything on our end, and we have fixed the bug in our projects that use a static library of our targets that made us introduce this problem in the first place. I feel good about these changes except-
> 
> What I am less in tune with is how to reproduce your original problem, to confirm that what I have done fixes this problem. I just understand how this should fix the problem _in theory_. How should I build with cmake? can I do this in the master branch or should I be doing this in a feature branch?
> 
> I have made the suggested changes [here ](https://github.com/Wenn0101/mbed-os/tree/cmake-ambiq-gcc-fix)to do some testing, but should these changes be added to [this PR](https://github.com/ARMmbed/mbed-os/pull/14042/)?

Hi @Wenchou01 
Thank you for assistance in this. Please use the branch in [the PR that mentions this issue](https://github.com/ARMmbed/mbed-os/pull/14042).
You can build by following the steps in https://github.com/hugueskamba/mbed-os/tree/hk_cmake_ambiq_micro_support/tools/cmake#how-to-build-an-application

I will port the changes you linked to my branch that is up for review in my PR.

Thanks

hugueskamba, 2020-12-17:
> Hi @Wenchou01
> Thank you for assistance in this. Please use the branch in [the PR that mentions this issue](https://github.com/ARMmbed/mbed-os/pull/14042).
> You can build by following the steps in [...]

Hi @Wenn0101,
I cherry picked your changes to my branch and I can confirm that the GCC_ARM toolchain error has been resolved!
Thank you.
I will now close this story as it has been successfully resolved.

closed 2020-12-17 by hugueskamba