ARMmbed/mbed-os #13999: Ticker IRQ triggering causes serial Rx IRQ to miss characters

pattyolanterns, 2020-12-02:
### Description of defect

Serial Rx IRQ missing groups of two characters.

String sent from host:
`1,1,c7,0,00c8,0,2,43013ae1`

String received by mcu:
`1,1,c7,0,00c8,0,2,430131` (ae) is missing.

I have seen loss of two characters in all indices of the string.

It appears that having multiple tickers defined elsewhere in the program is causing the missing characters when their IRQ is executed.

#### Target(s) affected by this defect ?
TARGET_STM32F411VE (using a custom board very similar to the NUCLEO_F411RE

#### Toolchain(s) (name and version) displaying this defect ?
GCC_ARM, version: 9 2020-q2-update

#### What version of Mbed-os are you using (tag or sha) ?

5.15.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli 1.10.5

#### How is this defect reproduced ? 

I have a thread running requesting serial data from an external host, once the data is received it requests for the next set of data.

I have also setup multiple tickers to poll different sensors, and debounce them with a timeout.

    void PollSensor::init() {
        _oldValue = _sensor.read();
        _pollTicker.attach_us(callback(this, &PollSensor::onTick), _frequency_ms * 1000);
    }

    void PollSensor::onTick() {
        _value = _sensor.read();

        if (_value != _oldValue) {
            if (_debounce_ms > 0) {
                _debounceTimeout.attach_us(callback(this, &PollSensor::onDebounce), _debounce_ms * 1000);
            }
            else {
                onDebounce();
            }
        }
    }

    void PollSensor::onDebounce() {
        uint8_t newValue = _sensor.read();
        if (newValue == _value) {
            _oldValue = _value;
            if (_value) {
                rise();
            } else {
                fall();
            }
        }
    }

the `rise` and `fall` methods are currently empty.

My `SerialRx` is as follows, where an rx isr adds characters to a `CircularBuffer`, to be consumed by the `lineThread`, where the line is added to an `mbed Mailbox`, which is then handled in a different consumer thread.

    SerialPacket* SerialRX::alloc() {
        return _mailManager.alloc();
    }

    void SerialRX::push(SerialPacket* packet) {
        _mailManager.push(packet);
    }

    void SerialRX::start(Callback<void(SerialPacket*)> func) {
        _mailManager.start(func);
        _serial->attach(callback(this, &SerialRX::handleChar), Serial::RxIrq);
        _thread.start(callback(this, &SerialRX::lineThread));
    }

    void SerialRX::handleChar() {
        char c = _serial->getc();
        // dont bother keeping a carriage return
        if (c != 0xD) {
            _buffer.push(c);
        }
    }

    void SerialRX::lineThread() {
        while (1) {
            SerialPacket* packet = alloc();

            if (packet != nullptr) {
                uint16_t str_i = 0;
                char c = '\0';
                packet->buffer[0] = c;

                while (c != '\n') {
                    while (_buffer.empty()) ThisThread::sleep_for(20);
                    _buffer.pop(c);
                    packet->buffer[str_i++] = c;
                }
                packet->buffer[str_i-1] = '\0';

                push(packet);
            }
        }
    }

By triggering the sensors while receiving data over serial, I am able to on occasion miss 2 characters (and always two characters). I have found it is faster to reproduce when adding more `PollSensor`'s (which use a ticker/timeout).

0xc0170, 2020-12-03:
cc @ARMmbed/team-st-mcd

kjbracey-arm, 2020-12-03:
Mbed OS already provides a buffered serial implementation that handles characters under intterupt and puts them in a buffer for a thread to read - no need to roll your own. (`UARTSerial` in Mbed OS 5.15, `BufferedSerial` in Mbed OS 6, IIRC).

Serial report reception is usually the first thing to go wrong if interrupt latency is poor.

Your serial RX handler has a flaw in that it assumes one character per interrupt. As load increases, there could be more than 1 available. Correcting this would somewhat increase the tolerance for interrupt latency.

I would really suggest using the well-proven buffered serial provided, but otherwise try a `while (_serial->readable())` loop in your IRQ handler.

There have been recent improvements to interrupt latency in the timing system, both general and ST-specific. I think Mbed OS 6 should perform better than 5.15. I forget which precise versions these hit - some are very recent.

pattyolanterns, 2020-12-03:
Hi, I appreciate the quick reply!

I have tried looping the ISR with `while (_serial->readable())`, and ran into the same issue.

I also implemented the UARTSerial in Mbed OS 5.15, and also ran into the same issue.

    void SerialRX::lineThread() {
        while (1) {
            SerialPacket* packet = alloc();

            if (packet != nullptr) {
                uint16_t str_i = 0;
                char buf[256];
                uint8_t lineRx = 0;

                while (!lineRx) { 
                    if (int num = _serial->read(buf, sizeof(buf))) {
                        for (int i = 0; i < num; i++) {
                            if (buf[i] == 0xD) continue;
                            if (buf[i] == '\n') {
                                packet->buffer[str_i] = '\0';
                                push(packet);
                                lineRx = 1;
                            }
                            else {
                                packet->buffer[str_i++] = buf[i];
                            }
                        }
                    }
                    ThisThread::sleep_for(20);
                }
            }
        }
    }

I will try porting over to mbed OS 6 to try it out the improved interrupt latency. But quick question before I do, are the `mbed_event_queue` `call_in` and `call_every` functions still susceptible to this same interrupt latency issue

pattyolanterns, 2021-01-31:
Just to close the loop on this for anyone who may stumble upon this thread.

Long story short I solved this problem by reducing my baud rate from 115200 to 38400.

My belief is that a group of tickers set up to fire on a steady interval (say 15 tickers every 100ms) would cause a decently large ISR queue to form. Where the RX handler was unable to handle the character in the buffer before it would be overwritten by the next character, due to the serial ISR handler being delayed by the many other ticker ISR handlers.

A more sustainable fix for this could be spacing out my polling tickers to they don't all fire at once, or using an `EventQueue` to remove the tickers out of ISR context.

closed 2021-01-31 by pattyolanterns