ARMmbed/mbed-os #14013: Porting a new BLE chip to Mbed-OS 6

ccchang12, 2020-12-08:
### Description of defect
We are considering to port our own BLE chip to Mbed-OS 6, and we have two questions now:

1. In latest BLE API, is it acceptable to use the Cordio stack + our own S/W LL+BB codes (include a library), and which without the WSF and PAL? This can help us a lot to port the BLE chip to Mbed-OS.

2. If Mbed-OS BLE supports Mesh in future. Does the Mesh is based on the Cordio stack, and all BLE chips can work well?

ccchang12, 2020-12-09:
This is a question report of porting a new BLE chip to Mbed, so these fields are None.

What target(s) are you using?
None

What toolchain(s) are you using?
None

It would help if you could also specify the versions of any tools you are using?
None

How can we reproduce your issue?
None

0xc0170, 2020-12-09:
Hi @ccchang12 , this is the issue tracker, questions should be asked on forums.mbed.com. I'll close this question.

cc @ARMmbed/mbed-os-pan

closed 2020-12-09 by 0xc0170