ARMmbed/mbed-os #14015: STM32F030: I2C slave data corruption

Arjun765, 2020-12-08:
### Description of defect
I'm trying to use a NUCLEO-F030R8 as an I2C slave, but when trying to read data, only the first read works that too is corrupt. I've tried using two masters, a NUCLEO-F767 and a Linux board. The master is sending 11 characters of "Hello World". I'm using the following code on the slave.

    #include "mbed.h"
    
    int main()
    {
            printf("=========================================================\r\n");
            DigitalOut led(LED1, false);
            I2CSlave slave(PB_9, PB_8);
            slave.address(0x84);
    
            uint32_t ctr = 0;
            while (1) {
                    char buffer[12]{};
                    const auto result = slave.receive();
    
                    if (result == I2CSlave::WriteAddressed) {
                            led = !led;
                            const auto len = slave.read(buffer, 11);
                            printf("%u:WA:%d:%s||", ++ctr, len, buffer);
                            for (const auto i : buffer) {
                                    printf("%x:", i);
                            }
                            puts("||\r\n");
                    }
            }
    }

I get the following output for the above code
`1:WA:0:�����@�����||90:cb:d8:d8:df:40:af:df:e4:d8:c8:0:||`

It seems all the bytes have been left shifted.

#### Target(s) affected by this defect ?
NUCLEO-F030R8

#### Toolchain(s) (name and version) displaying this defect ?
GCC 6.3.1

#### What version of Mbed-os are you using (tag or sha) ?
mbed-os-5.15.3

Arjun765, 2020-12-08:
[image]

LMESTM, 2020-12-08:
Hi - are there pull-up resistors on the I2C lines ? 
Missing pull-ups sometimes lead on unpredictable results ...

Arjun765, 2020-12-09:
Yes there are pullups. The exact same code above works fine on a NUCLEO-F767 board. It must be some bug in the F030 drivers.

Arjun765, 2020-12-09:
Wait, it seems the F767 also has this problem but its only triggered after about a minute.
`263:WA:0:�����@�����||90:cb:d8:d8:df:40:af:df:e4:d8:c8:0:||`
That is the last transaction that i received, after which it froze up

Arjun765, 2020-12-09:
I tried using a PI2 as I2C master and it works on it. The issue most likely is because my I2C master on the other Linux board is faulty or the pull up values are not correct.

closed 2020-12-09 by Arjun765