ARMmbed/mbed-os #14025: CMake Samsung: target files have dependency on connectivity

0xc0170, 2020-12-10:
### Description of defect

@ARMmbed/team-samsung Please review these files and why are they in 

targets/TARGET_Samsung/TARGET_SIDK_S5JS100/CMakeLists.txt - this is where we found these files:

```
# TODO CMAKE: The files commented below trigger a dependency on other libraries
# they should be moved somewhere in the connectivity directory.
        # dcxo_update.cpp
        # mbed_main_init.cpp
        # modem/modem_io_device.cpp
        # modem/modem_link_device_shmem.cpp
        # modem/s5js100_mbox_ipc.cpp
        # modem/shmem_save.cpp
```

You can try to add modem but it will request additional files that are not part of mbed-core. There are couple of files that include mbed.h , this should be removed (see mbed-os/targets/TARGET_Samsung/TARGET_SIDK_S5JS100/modem files).

#### Target(s) affected by this defect ?

SIDK_S5JS100

#### Toolchain(s) (name and version) displaying this defect ?

All

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-99.99.99

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

Not relevant

#### How is this defect reproduced ? 

Review the files shared above

pan-, 2020-12-10:
These files are not used at the moment, https://github.com/ARMmbed/mbed-os/pull/13814 seems to be the right place to address it.

zhizhezhu, 2020-12-11:
> These files are not used at the moment, #13814 seems to be the right place to address it.
@pan- 
Yes, #13814 is our target latest code. It can pass all test case and in order to get ARM Mbed Enable Certification.
I see #14024 is fork from ARMmbed:master. At present, the code of our target on the ARMmbed:master branch is incomplete.
So please based on #13814 to do CMake, If there is a problem with CMake on this PR, please let us know and we will solve it.
Thanks

0xc0170, 2020-12-11:
I merged 14024 that provides boilerplate for Samsung CMake. @zhizhezhu Please update 13814 and fix potential problems with the files mentioned here. If you got questions about CMake, ask here, we will assist.

zxytddd, 2020-12-18:
> I merged 14024 that provides boilerplate for Samsung CMake. @zhizhezhu Please update 13814 and fix potential problems with the files mentioned here. If you got questions about CMake, ask here, we will assist.

Hello, Martin,
As the PR 13814 is based from the previous mbed-os code without the CMake code.
How can I update the base code of PR 13814 and to make it has the PR 14024?
Can I cancel the 13814 and make a new PR with CMake update?

0xc0170, 2020-12-18:
If that works for you, yes you can create a new pull request or rebase the old one.

zhizhezhu, 2020-12-24:
Hi @0xc0170 
We had already add CMake in #13814 , please check.