ARMmbed/mbed-os #14026: SPI NAND Flash support

rogeryou, 2020-12-11:
### Description of defect
We cannot find SPI NAND Flash in any board supported by mbed. If we create SPI NAND Block device driver and commit the code. Will you consider merging the code? If so, how can you test the code without board containing the Flash?

And we also want to commit code to support our new kinds of Flash, like LybraFlash, RWW OctaFlash.
Besides, we want commit our middle ware named EEPROM emulator. Do you have any suggestions?
@jeromecoutant 

#### Target(s) affected by this defect ?

#### Toolchain(s) (name and version) displaying this defect ?

#### What version of Mbed-os are you using (tag or sha) ?

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

#### How is this defect reproduced ?

jeromecoutant, 2020-12-11:
Note I don't know any ST boards with SPI NAND Flash

0xc0170, 2020-12-16:
> We cannot find SPI NAND Flash in any board supported by mbed. If we create SPI NAND Block device driver and commit the code. Will you consider merging the code? If so, how can you test the code without board containing the Flash?

@ARMmbed/mbed-os-core Please review

@rogeryou As this is question, we can continue discussion on forum or just here, but I'll close this as it's not technical issue report.

closed 2020-12-16 by 0xc0170