ARMmbed/mbed-os #14039: STM32: Failure in CAN initialization leads to system crash

FabianInostroza, 2020-12-13:
### Description of defect
If you use CAN bus on a stm32 target and the transceiver or network is bad, the system crashes and it's unusable.
Even your application can't report that something is bad with the CAN network.

#### Target(s) affected by this defect ?
At least STM32

#### Toolchain(s) (name and version) displaying this defect ?
gcc version 9.3.1 20200408 (release) (GNU Arm Embedded Toolchain 9-2020-q2-update) 

#### What version of Mbed-os are you using (tag or sha) ?
mbed-os-5.15.5

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli 1.10.5

#### How is this defect reproduced ? 
To reproduce the error on boot, connect CANH to +5v and CANL to gnd.
An alternative is to use the following code:
```
    DigitalOut can1_tx(CAN1_TX, 0);
    can1.frequency(250000);
```
can1 is the CAN object initialized using CAN1_TX pin as td pin, frequency is irrelevant. The system will crash on the frequency() call.

0xc0170, 2021-01-11:
cc @ARMmbed/team-st-mcd

jeromecoutant, 2021-06-22:
Hi @FabianInostroza 
Do you have the chance to try latest mbed-os label (6.x) ?
Thx

jeromecoutant, 2021-06-22:
> the system crashes and it's unusable.

Could you add more details ?

closed 2021-10-20 by ciarmcom