ARMmbed/mbed-os #14052: UnbufferedSerial does not expose enable_* functions from SerialBase; behavior not predictable when exposed

petertorelli, 2020-12-15:
### Description of defect

The documentation claims `UnbufferedSerial` has two functions, `enable_input` and `enable_output`. However, they are not defined in the base class and SerialBase is not accessible.

#### Target(s) affected by this defect ?

I tested on a CY8CPROTO_062_4343W and NUCLEO_F429ZI, but I assume all since it is a header issue.

#### Toolchain(s) (name and version) displaying this defect ?

GCC_ARM

#### What version of Mbed-os are you using (tag or sha) ?

6.5.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

mbed: 1.10.4

#### How is this defect reproduced ? 

~~~
#include "mbed.h"

UnbufferedSerial pc(USBTX, USBRX);

int
main()
{
	printf("Serial enabled, disabling.\n");
	pc.sync();
	pc.enable_input(false);
	pc.enable_output(false);
	pc.enable_input(true);
	pc.enable_output(true);
	printf("Serial enabled, this should be visible.\n");

}
~~~

This will fail to compile with this error

~~~
./main.cpp:13:5: error: request for member 'enable_output' is ambiguous
  pc.enable_output(true);
     ^~~~~~~~~~~~~
In file included from ./mbed-os/connectivity/netsocket/include/netsocket/nsapi_ppp.h:22,
                 from ./mbed-os/mbed.h:28,
                 from ./main.cpp:1:
./mbed-os/platform/include/platform/FileHandle.h:210:17: note: candidates are: 'virtual int mbed::FileHandle::enable_output(bool)'
     virtual int enable_output(bool enabled)
                 ^~~~~~~~~~~~~
In file included from ./mbed-os/drivers/include/drivers/UnbufferedSerial.h:26,
                 from ./mbed-os/mbed.h:71,
                 from ./main.cpp:1:
./mbed-os/drivers/include/drivers/SerialBase.h:144:10: note:                 'void mbed::SerialBase::enable_output(bool)'
     void enable_output(bool enable = true);
          ^~~~~~~~~~~~~
~~~

It can be fixed by adding these two lines to `UnbufferedSerial.h`:

~~~
    using SerialBase::enable_input;
    using SerialBase::enable_output;
~~~

Furthermore, this code works with the fixes on the NUCLEO, but it fails on the CYPRESS board (screen fills with garbage characters). I suspect the deinit/reinit of the serial object is the culprit in the Cypress case.

0xc0170, 2021-01-11:
@ARMmbed/mbed-os-hal

ithinuel, 2021-01-14:
This is higher level abstraction than the HAL.

@kjbracey-arm for `FileHandle::enable_input` and @hugueskamba for `UnbufferedSerial`.

kjbracey-arm, 2021-01-14:
> running on a Cypress board corrupts the port.

It looks like you're running two drivers on the port.  - your `pc` and the C library stdout. 

That's likely to cause problems, not directly connected to the enable calls. You need to have only one driver.

Either arrange for the C library to use your `pc` rather than its own driver by defining `mbed_override_console`, or get access to the C library's driver by calling `mbed_file_handle(STDOUT_FILENO)`. Both are in mbed_retarget.h.

eg

```
#include "mbed.h"

int
main()
{
        // both are probably the same, but let's be picky...
        FileHandle *console_in = mbed_file_handle(STDIN_FILENO);
        FileHandle *console_out = mbed_file_handle(STDOUT_FILENO);
	printf("Serial enabled, disabling.\n");
	console_out->sync();
	console_in->enable_input(false);
	console_out->enable_output(false);
	console_in->enable_input(true);
	console_out->enable_output(true);
	printf("Serial enabled, this should be visible.\n");

}
```

That would sidestep the `UnbufferedSerial` problem, which still needs fixing.

kjbracey-arm, 2021-01-14:
Actually my code above quite possibly doesn't work, if the console is unbuffered, because it seems likely that the same thing causing an ambiguity error would mean `UnbufferedSerial` hasn't overridden the default "just return error" implementation of the virtual method `FileHandle::enable_input`.

I think the `using SerialBase::enable_input` suggested above would sort that out too, but I'm not totally sure.

ithinuel, 2021-01-14:
Is that the gist of the issue we're seeing : https://godbolt.org/z/aoE78x ?

kjbracey-arm, 2021-01-15:
Interestingly, it looks like the `using` isn't sufficient to perform the virtual override there. Most curious. I've failed to find anything clear on the topic.

0xc0170, 2021-01-15:
> This is higher level abstraction than the HAL.

We need to clear this as previously hal team was also responsible for drivers (the higher abstraction).

petertorelli, 2021-01-15:
Hi @kjbracey-arm,

Good point, about the "two drivers" issue: I had assumed that creating a driver on USBTX/USBRX would give access to the configuration of the DEFAULT output driver. (Why? Because this was a trick in the old mbed versions to change the default baud rate: instantiate a port `Serail pc(USBTX, USBRX);` and then use `pc.baud(...)`.) That could very well be the root of the functional problem, but it still leaves the inconsistency between the documentation and the compile errors. Thanks!

Peter

kjbracey-arm, 2021-01-18:
That trick was indeed supported, and maybe still is as long as the console is unbuffered. It requires some horrible code in the C library retargetting and the HAL to spot it happening, intercept the operation, and "do what they want rather than what they're saying".  But I don't believe that trick works any more once the more complicated buffered serial driver is in use for the console. (Is that the default yet?)  I don't know whether it's your issue though.

I'm sure a various bits of documentation and examples have not been updated with new guidance. The main reason for the original trick was that there was no config option to specify console parameters, so that trick was the _only_ way to configure the console.

Proper config options such as `platform.stdio-baud-rate` were added a long time ago, rendering that basic concept unnecessary. And `mbed_file_handle` was added more recently to give nice run-time access for those needing it.

petertorelli, 2021-01-20:
Hi @kjbracey-arm,

What I'm trying to do here is drive down the Cypress PSOC6 power. According to the mbed docs, I cannot enter the lowest sleep state because the UARTs need their I/O disabled (this is in the header file `UnbufferedSerial.h`). Even if I manage to replace redirect vsprintf's use of putc with my own UnbufferedSerial.write(), I cannot find a way to disable the default serial port I/O so that it powers down. Or at least, I can't seem to do that. Is there a direct hardware interface to the default serial port driver? Your previous post seems to indicate there is not, since changing the baud requires a config option. Or should this be in a different ticket?

Thanks,
Peter

PR #14175 (2021-01-25): Add declaration to expose enable_* functions from SerialBase

### Summary of changes 

Fixes #14052 

UnBufferedSerial is missing a declaration to expose enable_input and enable_output, which are inherited from the private base class SerialBase. Add the using-declaration to the class definition to expose these functions. 

#### Impact of changes 

None
#### Migration actions required
closed 2021-01-25 by 0xc0170