ARMmbed/mbed-os #14053: BLE scanning does not work when extended advertising is disabled

AGlass0fMilk, 2020-12-16:
### Description of defect

When the BLE stack is configured to disable extended advertising as below:

```
"ble.ble-feature-periodic-advertising": false,
"ble.ble-feature-extended-advertising": false
```

Scanning behavior is unexpectedly affected. Specifically, after initiating scanning, a registered `Gap::EventHandler`'s `onAdvertisingReport` callback will never be executed.

I have tested this behavior on the nRF52832 so far.

#### Target(s) affected by this defect ?

NRF52_DK

Possibly NRF52840_DK and other BLE targets.

#### Toolchain(s) (name and version) displaying this defect ?
```
arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 9-2019-q4-major) 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]
```

#### What version of Mbed-os are you using (tag or sha) ?

d6784c3ee6ada8e886801d0197904d3ab94c891c

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
```
 pip list
Package             Version     Location
------------------- ----------- -------------------------------------------------------------------
appdirs             1.4.3
asn1ate             0.6.0
attrs               19.3.0
Automat             20.2.0
beautifulsoup4      4.6.3
ble-serial          1.2.1
bleak               0.9.1
bluepy              1.3.0
cbor                1.0.0
certifi             2019.11.28
cffi                1.14.1
chardet             3.0.4
Click               7.0
cmsis-pack-manager  0.2.10
cobs                1.1.4
colorama            0.3.9
constantly          15.1.0
crc16               0.1.1
crccheck            0.6
cryptography        2.9.2
cycler              0.10.0
docopt              0.6.2
ecdsa               0.15
elftools            0.1.0.dev0
fasteners           0.15
Flask               1.1.2
future              0.16.0
futures             3.1.1
fuzzywuzzy          0.18.0
grip                4.5.2
hidapi              0.9.0.post2
hyperlink           20.0.1
icetea              1.2.4
idna                2.7
imgtool             1.7.0rc1
importlib-metadata  1.6.0
incremental         17.5.0
iniconfig           1.1.1
intelhex            2.2.1
itsdangerous        1.1.0
Jinja2              2.10.3
jsonmerge           1.7.0
jsonschema          2.6.0
junit-xml           1.8
kiwisolver          1.2.0
lockfile            0.12.2
Logbook             1.5.3
manifest-tool       1.5.2
Markdown            3.3.2
MarkupSafe          1.1.1
matplotlib          3.3.0
mbed-ble-test-suite 0.0.1       /home/gdbeckstein/Documents/mbed-os-bluetooth-integration-testsuite
mbed-cli            1.10.4
mbed-cloud-sdk      2.0.8
mbed-flasher        0.10.1
mbed-greentea       1.7.4
mbed-host-tests     1.5.10
mbed-ls             1.7.12
mbed-os-tools       0.0.15
milksnake           0.1.5
monotonic           1.5
numpy               1.19.1
packaging           20.4
path-and-address    2.0.1
pc-ble-driver-py    0.14.2
Pillow              7.2.0
pip                 20.2
pkg-resources       0.0.0
pluggy              0.13.1
prettytable         0.7.2
protobuf            3.5.2.post1
psutil              5.6.6
py                  1.9.0
pyasn1              0.2.3
pycparser           2.20
pycryptodome        3.9.8
pyelftools          0.25
Pygments            2.7.1
PyHamcrest          2.0.2
pyparsing           2.4.7
pyrsistent          0.16.0
pyserial            3.4
pytest              6.1.1
python-dateutil     2.8.1
python-dotenv       0.14.0
pyusb               1.0.2
PyYAML              4.2b1
requests            2.20.1
semver              2.10.2
setuptools          46.1.3
six                 1.12.0
soupsieve           2.0
toml                0.10.1
trollius            2.1.post2
Twisted             20.3.0
txdbus              1.1.2
urllib3             1.24.2
Werkzeug            1.0.1
wheel               0.34.2
wrapt               1.12.1
yattag              1.13.2
zipp                3.1.0
zope.interface      5.2.0

```

#### How is this defect reproduced ? 

Add the following configuration options to a BLE application that performs **basic** ("classic" I suppose?) scanning (eg: https://github.com/ARMmbed/mbed-os-example-ble/tree/master/BLE_GAP):

```
"ble.ble-feature-periodic-advertising": false,
"ble.ble-feature-extended-advertising": false
```

paul-szczepanek-arm, 2021-01-04:
Thanks. The problem was the use of extended HCI commands even when the feature is disabled.
https://github.com/ARMmbed/mbed-os/pull/14111
This was causing the controller to reject legacy commands later on.

AGlass0fMilk, 2021-01-05:
Should be fixed by #14111. I will test and reopen the issue if I encounter any other problems.

closed 2021-01-05 by AGlass0fMilk