ARMmbed/mbed-os #14058: STM32F7 flash driver disables interrupts unnecessarily

kjbracey-arm, 2020-12-16:
### Description of defect

The STM32F7 flash driver disables interrupts during erase operations (and maybe programming). As erase operations can take hundreds of milliseconds this can cause major system disruption.

The STM32F7 has dual-bank flash, so it is possible to configure the system so that one bank is used for the program and one bank used for other storage such as KVStore. Configured like this there should be no impact on the system when erasing or writing to KVStore.

However, even if configured for dual-bank, the interrupt disable happens for the entire flash write, which could be 800ms.

If configured for dual-bank, the STM32F7 driver should only disable interrupts if performing a flash operation on a bank that the program is in. (This is possibly easier to express in English than as a C check though). It is possible that no interrupt disables are needed at all - if operating on the current bank, I believe the system will automatically stall the CPU, except for certain fast-programming operations that would need to be executed from RAM anyway that we're not using.

Issue originally noted in https://github.com/ARMmbed/mbed-os/issues/13772 - that was marked resolved in that the system timer was made more robust to cope better after the long interrupt blockage - so it didn't cause ongoing problems - but it didn't resolve the underlying platform problem that makes the unnecessary blockage occur.

#### Target(s) affected by this defect ?

STM32F7 family

#### Toolchain(s) (name and version) displaying this defect ?

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.6.0-rc1

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

#### How is this defect reproduced ? 

Set up a `Timer` to blink an LED at, say, 4Hz, while running any KVStore test, such as the test program in #13772 on a system configured for dual bank operation. The timer should remain regular during flash operations, but given the current long interrupt disables would not.

jeromecoutant, 2021-03-24:
@kjbracey-arm do you suggest to remove the critical sections ?
(added for #13792)

kjbracey-arm, 2021-03-24:
Hiya.

Total removal may be an option, but I'm not sure. Relevant paragraph from description:

> If configured for dual-bank, the STM32F7 driver should only disable interrupts if performing a flash operation on a bank that the program is in. (This is possibly easier to express in English than [...]

So it mustn't disable interrupts when programming a non-program bank. Maybe you don't need to disable interrupts at all - you can probably figure that out better than me. That would be easier than figuring out how to detect "programming a non-program bank".

jeromecoutant, 2021-03-24:
Mmm
As #13792 request was initially only for erase issues,
and got issue also with STM32L5 with these new critical sections,

I propose to remove the critical section in the program functions,
and keep only for the moment the one in erase functions.

jeromecoutant, 2021-03-24:
@moshe-shahar

moshe-shahar, 2021-03-24:
@jeromecoutant , I can't give a good answer for it at that point.
I run into that during erase but as it seems, it can be raised during program too.