ARMmbed/mbed-os #14059: STM32 I2C libraries discard I2C error cause

More-Wrong, 2020-12-16:
### Description of defect

The error handler code in the I2C system (in `i2c_api.c`), for the STM devices discards the error status returned by the HAL, only printing it, without actually setting the I2C status. It sets `obj_s->event = I2C_EVENT_ERROR;` having ignored `hi2c->ErrorCode`, which contains the cause of the failure. 
This results in the I2C system being reset, even if the only error was a `NACK` response, and means the calling code cannot tell what caused the error.

I will happily write a fix for this (particularly as I need this functionality anyway), but I only have the one device to test it on, and would like to know if there is a good reason it works the way it does, before I start.

#### Target(s) affected by this defect ?

All of the STM family, although I only have an STM32G484RE to test on.
 
#### Toolchain(s) (name and version) displaying this defect ?

GCC_ARM, although the issue will affect all toolchains.

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.6.0 and mbed-os-6.5.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
n/a

#### How is this defect reproduced ? 

The defect can be seen in the code, but can be reproduced by running a disconnected I2C system, and printing the error response. Although it should give an `I2C_EVENT_ERROR_NO_SLAVE`, it only gives an `I2C_EVENT_ERROR`. 

From what I can tell by looking at other targets, it is meant to give `I2C_EVENT_ERROR_NO_SLAVE | I2C_EVENT_ERROR`, is this correct?

More-Wrong, 2020-12-16:
Looking further at the API for the I2C system, the blocking methods (read, write, etc.) provide no feedback on the reason for failure, only the non-blocking system has any way of getting this information out. This seems odd, as the difference between an address NACK (implies the other end doesn't exist) and a data NACK (implies the other end is too busy) is huge.

0xc0170, 2021-01-11:
cc @ARMmbed/team-st-mcd

jeromecoutant, 2021-02-24:
Hi

> I will happily write a fix for this (particularly as I need this functionality anyway), but I only have the one device to test it on, and would like to know if there is a good reason it works the way it does, before I start.

No, there is no good reason. You can provide some fix tested on your target.
I will test on other STM32.

jeromecoutant, 2021-05-06:
@More-Wrong Please check #14631

jeromecoutant, 2021-05-10:
As #14631 is merged, maybe we can close this one ?
Thx