ARMmbed/mbed-os #14069: [mbed-os-6.6.0] minimal-printf incorrect output

caoddx, 2020-12-18:
### Description of defect

Floating point number is not printed correctly.

test code
```c
  printf("123.456: %f\n", 123.456);
  printf("0.456: %f\n", 0.456);
  printf("-0.456: %f\n", -0.456);
  printf("123.0: %f\n", 123.0);

  // output:
  // 123.456: 123.000123
  // 0.456: 0.000000
  // -0.456: -0.000000
  // 123.0: 123.000123
```

mbed_app.json
```json
{
    "target_overrides": {
        "*": {
            "target.printf_lib": "minimal-printf",
            "platform.minimal-printf-enable-floating-point": true,
            "platform.minimal-printf-set-floating-point-max-decimals": 6,
            "platform.minimal-printf-enable-64-bit": true
        }
    }
}
```

It seems that the variable `precision` in the file [mbed_printf_implementation.c](https://github.com/ARMmbed/mbed-os/blob/d6784c3ee6ada8e886801d0197904d3ab94c891c/platform/source/minimal-printf/mbed_printf_implementation.c#L299) is not used.

#### Target(s) affected by this defect ?

should be all

#### Toolchain(s) (name and version) displaying this defect ?

GCC_ARM

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.6.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

mbed-cli

#### How is this defect reproduced ? 

use the above test code and mbed_app.json

0xc0170, 2020-12-18:
This PR might have caused this https://github.com/ARMmbed/mbed-os/pull/13548. Have you tested 6.5 version if it works as expected?

cc @EmbedEngineer

EmbedEngineer, 2020-12-18:
@0xc0170 yes this has been tested by the green tea tests and there it works. Just take a look at the appropriate file.

helpfulchicken, 2021-01-11:
I have come across this problem myself and found the fix.
This line: 
https://github.com/ARMmbed/mbed-os/blob/e917282d6072e5bd82cc747954bc31189108ac79/platform/source/minimal-printf/mbed_printf_implementation.c#L310

should be 
`MBED_UNSIGNED_STORAGE decimal = value * precision;`

caoddx, 2021-01-11:
@helpfulchicken 
Have you tested my test case?

Here's my solution:

```c++
...
/* get decimal part */
double precision = 1.0;

for (size_t index = 0; index < dec_precision; index++) {
    precision *= 10;
}

value -= integer; // <---------------- add code
/* convert to positive number */
if (value < 0.0) {
    value *= -1.0;
}

value *= precision; // <---------------- add code
MBED_UNSIGNED_STORAGE decimal = value;

/* round up or down */
value -= decimal;

if (!((value > -0.5) && (value < 0.5))) {
    decimal++;
}
...
```

0xc0170, 2021-01-11:
@caoddx please create a new PR with a fix (also a test case to have a failure fixed).

helpfulchicken, 2021-01-17:
> I have come across this problem myself and found the fix.
> This line:
> https://github.com/ARMmbed/mbed-os/blob/e917282d6072e5bd82cc747954bc31189108ac79/platform/source/minimal-printf/mbed_printf_implementation.c#L310
> 
> should be
> `MBED_UNSIGNED_STORAGE decimal = value * precision;`
> 
> Otherwise the assignment truncates all decimal information, which completely spoils the intended behaviour.

I have realised I was mistaken in my fix. 
It should be 
`MBED_UNSIGNED_STORAGE decimal = (value - integer) * precision;`

LDong-Arm, 2021-01-20:
@helpfulchicken @caoddx Please create a PR with your fix, and we'll review it and get it in. Thanks!

@evedon FYI

PR #14179 (2021-02-05): Fix minimal-printf floating point decimal output error

### Summary of changes 

Fix #14069

Need this configuration file for testing: `platform/tests/TESTS/mbed_platform/minimal-printf/compliance/test_config.json`
closed 2021-02-05 by 0xc0170