ARMmbed/mbed-os #14070: USB Driver: Build Error Because of Missing Include Folder in CMakeLists.txt

ahmetalincak, 2020-12-18:
### Description of defect

USB Driver can't be built because of missing headers. I loaded `mbed-usb` and `mbed-events` from my root _CMakeLists.txt_ and compilation failed due to not finding header files.

This can be resolved by adding `include/usb/internal` line to `mbedos/drivers/usb/CMakeLists.txt`.

#### Target(s) affected by this defect ?
All targets which have USBDevice attirubute.

#### Toolchain(s) (name and version) displaying this defect ?
GCC ARM 9 2020-q2-update

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os v6.6

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli 2 v4.0.0

#### How is this defect reproduced ? 
Update your root CMakeLists.txt file as following;

`
target_link_libraries(${APP_TARGET}
    PRIVATE
        mbed-os
        mbed-usb
        mbed-events 
)
`

And run below command

`mbed-tools compile -t GCC_ARM -m K64F`

In my case I got failure for MAX32630FTHR board. Please see attachment.

[image]

ahmetalincak, 2020-12-18:
#14071 opened for this issue.

0xc0170, 2020-12-18:
Thanks for the fix, I'll close as resolved.

closed 2020-12-18 by 0xc0170