ARMmbed/mbed-os #14081: ble in mbed os has a bug when a the mobile phone sleeps, the embedded device does not reset the connection

sherifomran, 2020-12-22:
Describe the bug

A clear and concise description of what the bug is.
the bluetooth driver has a bug exactly when a device (ios) connects to the peripheral (nordic chip) and the device sleeps and causing the connection to disconnect from one side only, the peripheral can not detect this disconnection and must get a hardware reset in order to be able to connect to it again.
It should detect the disconnection using a time or indication or something else.

To Reproduce
connect ios to peripheral and let it sleep. After some minuites it is disconnected from one side, try reconnection, it will fail.

Expected behavior
Reconnection successful

Screenshots

Context (please complete the following information):

    Mbed OS version: [e.g. 6.4]
    Compiler: [e.g. GCC, ARM, ..]
    Target [e.g. NRF52840_DK]

Additional context
Add any other context about the problem here.

jrobeson, 2020-12-22:
@sherifomran : you still have to fill out those fields you did not fill out (mbed us version, compiler, target) if you want the bug looked at properly.

0xc0170, 2021-01-11:
As stated earlier, I'll close this as it does not contain required information. Please reopen with an update.

closed 2021-01-11 by 0xc0170