ARMmbed/mbed-os #14082: STM32L073RZ without LSE

ewhyte4345, 2020-12-22:
### Description of defect

Very similar issue to https://github.com/ARMmbed/mbed-os/issues/13127

I want to run the Nucleo-L073RZ just from the internal oscillators, so I can remove the 32kHz LSE crystal.

Running the standard mbed OS blinky example with R34 and R36 removed I get the following terminal output:

++ MbedOS Error Info ++
Error Status: 0x80FF0100 Code: 256 Module: 255
Error Message: Fatal Run-time error
Location: 0x80048D1
Error Value: 0x0
Current Thread: Id: 0x0 Entry: 0x8000215 StackSize: 0x0 StackMem: 0x0 SP: 0x20004E7C
For more info, visit: https://mbed.com/s/error?error=0x80FF0100&tgt=NUCLEO_L073RZ
– MbedOS Error Info –
HAL_RCC_OscConfig ERROR

When the same program is compiled for Nucleo-F091RC (with R34 and R36 also removed) the program runs fine.

#### Target(s) affected by this defect ?
NUCELO-L073

#### Toolchain(s) (name and version) displaying this defect ?
mbed online compiler

#### What version of Mbed-os are you using (tag or sha) ?
mbed-os-6.6.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed online compiler

#### How is this defect reproduced ? 
Run mbed-os-blinky on NUCLEO--L073 with R34 and R36 removed

ewhyte4345, 2020-12-22:
The issue can be resolved by putting the following in an mbed_app.json file:

{
"target_overrides":
    {
"*":
        {
"target.lse_available": "0"
        }
    }
}

It would be nice to improve the documentation around this or modify the mbed library so that the above fix is not necessary.

jeromecoutant, 2020-12-22:
> It would be nice to ... modify the mbed library so that the above fix is not necessary.

What do you suggest ?

ewhyte4345, 2020-12-22:
Hi Jerome,

Sorry, I do not have a deep enough knowledge of the platform to make an accurate suggestion, but it would be better if the library supported the NUCLEO-L073 board without 32kHz xtal (as it does for F091). I suppose if it doesn't find an LSE clock it should revert to LSI automatically.

0xc0170, 2021-01-11:
@jeromecoutant should a doc be updated to resolve this?

PR #14338 (2021-03-03): STM32: update readme with clock information
### Summary of changes 

Quick addition in the STM32 readme file about clock settings

fix #14082 

#### Impact of changes 

#### Migration actions required 

### Documentation 

----------------------------------------------------------------------------------------------------------------
### Pull request type
closed 2021-03-03 by 0xc0170