ARMmbed/mbed-os #14097: USBSerial printf() HardFault only on Linux with GCC_ARM

IanBurwell, 2020-12-26:
### Description of defect
Simply put, when I compile on windows my code run as expected, but when I do so on Linux with the same code and mbed-os I get a HardFault (see below). The issue seems to stem when I call `usbserial.printf("Hello World\n");` as without it my code (see below) works fine. 

A point of note is that I have been using a custom target which I put into a modified mbed-os. This target (NU32) is a minimal STM32RE401RE board with an 8MHz clock. If anyone is interested in the structure [here is my repo](https://github.com/IanBurwell/NUice/tree/mbed/firmware/mbed). I am uploading via DFU and debugging via UART and an FTDI adapter.

main.cpp
```cpp
#include "mbed.h"
#include "platform/mbed_thread.h"
#include "USBSerial.h"

// Blinking rate in milliseconds
#define BLINKING_RATE_MS    500
USBSerial usbserial;

int main()
{
    // Initialise the digital pin LED1 as an output
    DigitalOut led(PA_9);
    DigitalOut led2(PA_10);
    //AnalogIn   ain(PA_0);
    
    led2 = !led;
    while (true) {
        led = !led;
        led2 = !led2;
       
        usbserial.printf("Hello World\n");
       
        thread_sleep_for(BLINKING_RATE_MS);
    }
}
```

HardFault
```
++ MbedOS Fault Handler ++

FaultType: HardFault

Context:
R0: 0
R1: 0
R2: 800274D
R3: 0
R4: 0
R5: 200018C4
R6: 1
R7: 0
R8: 1
R9: 50000920
R10: 50000000
R11: 1
R12: 20001C6D
SP   : 20017F50
LR   : 8002757
PC   : 0
xPSR : 40000053
PSP  : 20002B90
MSP  : 20017F30
CPUID: 410FC241
HFSR : 40000000
MMFSR: 0
BFSR : 0
UFSR : 2
DFSR : 0
AFSR : 0
Mode : Handler
Priv : Privileged
Stack: MSP

-- MbedOS Fault Handler --

++ MbedOS Error Info ++
Error Status: 0x80FF013D Code: 317 Module: 255
Error Message: Fault exception
Location: 0x0
Error Value: 0x20002DB8
Current Thread: main <handler> Id: 0x20001294 Entry: 0x8001D89 StackSize: 0x1000 StackMem: 0x20001DB8 SP: 0x20017F50
For more info, visit: https://mbed.com/s/error?error=0x80FF013D&tgt=NU32
-- MbedOS Error Info --
```

Linux Build
```bash
$ mbed compile -m NU32 -t GCC_ARM --source nuice-example --source mbed-os --build BUILD/nuice-example
[mbed] Working path "/home/en/NUice/firmware/mbed" (library)
[mbed] Program path "/home/en/NUice/firmware/mbed"
Building project nuice-example (NU32, GCC_ARM)
Scan: nuice-example
Scan: mbed-os
Link: nuice-example
Elf2Bin: nuice-example
| Module               |     .text |    .data |     .bss |
|----------------------|-----------|----------|----------|
| [fill]               |   140(+0) |    7(+0) |   20(+0) |
| [lib]/c.a            |  8300(+0) | 2108(+0) |   89(+0) |
| [lib]/gcc.a          |   808(+0) |    0(+0) |    0(+0) |
| [lib]/misc           |   188(+0) |    4(+0) |   28(+0) |
| mbed-os/cmsis        |  7458(+0) |  168(+0) | 5957(+0) |
| mbed-os/connectivity |   124(+0) |    0(+0) |    0(+0) |
| mbed-os/drivers      |  9118(+0) |    0(+0) |    2(+0) |
| mbed-os/hal          |  1430(+0) |    4(+0) |   66(+0) |
| mbed-os/platform     |  5752(+0) |  260(+0) |  436(+0) |
| mbed-os/rtos         |   398(+0) |    0(+0) |    0(+0) |
| mbed-os/targets      | 12496(+0) |    9(+0) | 1430(+0) |
| nuice-example/main.o |   184(+0) |    0(+0) | 1140(+0) |
| Subtotals            | 46396(+0) | 2560(+0) | 9168(+0) |
Total Static RAM memory (data + bss): 11728(+0) bytes
Total Flash memory (text + data): 48956(+0) bytes

Image: BUILD/nuice-example/nuice-example.bin
```

Windows Build
```bash
> mbed compile -m NU32 -t GCC_ARM --source nuice-example --source mbed-os --build BUILD/nuice-example
[mbed] Working path "C:\Users\spygu\Desktop\nuice\firmware\mbed" (library)
[mbed] Program path "C:\Users\spygu\Desktop\nuice\firmware\mbed"
[Warning] @,: Compiler version mismatch: Have 10.2.1; expected version >= 9.0.0 and < 10.0.0
Building project nuice-example (NU32, GCC_ARM)
Scan: nuice-example
Scan: mbed-os
Compile [ 99.2%]: mbed_rtx_handlers.c
Compile [ 99.4%]: mbed_rtx_idle.cpp
Compile [ 99.5%]: mbed_error.c
Compile [ 99.6%]: FATFileSystem.cpp
Compile [ 99.7%]: LittleFileSystem.cpp
Compile [ 99.8%]: kv_config.cpp
Compile [ 99.9%]: PlatformStorage.cpp
Compile [100.0%]: main.cpp
Link: nuice-example
Elf2Bin: nuice-example
| Module               |     .text |    .data |     .bss |
|----------------------|-----------|----------|----------|
| [fill]               |   162(+0) |    7(+0) |   23(+0) |
| [lib]\c.a            |  8576(+0) | 2108(+0) |   58(+0) |
| [lib]\gcc.a          |   772(+0) |    0(+0) |    0(+0) |
| [lib]\misc           |   188(+0) |    4(+0) |   28(+0) |
| mbed-os\cmsis        |  7372(+0) |  168(+0) | 5953(+0) |
| mbed-os\connectivity |   124(+0) |    0(+0) |    0(+0) |
| mbed-os\drivers      |  8984(+0) |    0(+0) |    2(+0) |
| mbed-os\hal          |  1430(+0) |    4(+0) |   66(+0) |
| mbed-os\platform     |  5730(+0) |  260(+0) |  436(+0) |
| mbed-os\rtos         |   374(+0) |    0(+0) |    0(+0) |
| mbed-os\targets      | 12348(+0) |    9(+0) | 1430(+0) |
| nuice-example\main.o |   184(+0) |    0(+0) | 1140(+0) |
| Subtotals            | 46244(+0) | 2560(+0) | 9136(+0) |
Total Static RAM memory (data + bss): 11696(+0) bytes
Total Flash memory (text + data): 48804(+0) bytes

Image: BUILD/nuice-example\nuice-example.bin
```

#### Target(s) affected by this defect ?

- NU32 (Custom STM32401RE implementation)

#### Toolchain(s) (name and version) displaying this defect ?

- GCC_ARM 9

#### What version of Mbed-os are you using (tag or sha) ?

- mbed-os-6.4.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

- mbed-cli v0.4.10
- STM32CubeProgrammer
- Ubuntu 18.04
- Windows 10

#### How is this defect reproduced ? 
As it is a custom target it might be hard for one to reproduce. I currently do not have a Nucleo 401RE but presume that a similar error would occur on that target.

0xc0170, 2021-01-11:
Have you compared the images between linux and windows? What is causing the Hardfault - code line ? is it always the same line?

cc @ARMmbed/team-st-mcd

IanBurwell, 2021-01-12:
I have not compared the images because I do not have the know-how to do so. Regarding the line that causes the Hardfault, I only have the ability to debug over UART and have provided the error from that. I do not have access to a SWD or JTAG adapter at the moment so that's the best I can do.

AGlass0fMilk, 2021-01-14:
If you can post the binaries built on both Windows and Linux I'll do a diff and see if it's different.

Are you using GCC ARM to build on Windows _and_ Linux, both?

jeromecoutant, 2021-04-27:
Hi
Is this issue still an issue?
Did you check with the latest release ?
Thx

IanBurwell, 2021-05-09:
Ah, sorry. I have not investigated too much as I needed a working system ASAP and went with another solution. I do not intend to dig into this much further.

closed 2021-05-09 by IanBurwell