ARMmbed/mbed-os #14106: Fail compilation if using obsolete compilers (GCC6).

teetak01, 2020-12-30:
### Description of defect

Instead of warning, Mbed OS should prevent compilation with very obsolete compiler versions.

Specifically GCC cross-compiler 6 has still been used by engineers to compile Mbed OS applications and with Mbed OS 6 the GCC6 code will not either compile, or will fail very unexpectedly at runtime.

I would suggest changing the:

[Warning] @,: Compiler version mismatch: Have 6.3.1; expected version >= 9.0.0 and < 10.0.0

to

[Error] @,: Compiler version mismatch: Have 6.3.1; expected version >= 9.0.0 and < 10.0.0

I do not think 7/8 was ever supported, or if those should be treated separately, but it would be better to start giving hard failures for compiler versions that have not been supported for long time anymore. Also GCC6 specifically has been unsupported since 5.15 (last supported for 5.14 release).

I have seen people still getting very unexpected failures/crashes or even compilation issues with GCC6 when combined with latest Mbed OS releases. It would save some debugging time to just hard-fail instead. People might not tend to realize the compiler might be causing issues, even if there is a warning.

#### Target(s) affected by this defect ?

#### Toolchain(s) (name and version) displaying this defect ?

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.5.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

#### How is this defect reproduced ?

0xc0170, 2021-01-04:
Thanks for the report. An error would make sense as even if it compiles, there could be hard-to-find errors.

teetak01, 2021-03-03:
Maybe same thing also for ARMCC 6.11 or older. Managed to do that mistake myself. Also related to #12764