ARMmbed/mbed-os #14114: Serial.h file not found problem

hannbusann, 2021-01-05:
Hi! I use the default blinky example to start my project, but will I use：
```cpp
#include <Serial.h>
Serial pc(PA_9,PA_10);
```
The compiler output error that "Serial.h" file not found. What's wrong with my configuration or code? Please.

hannbusann, 2021-01-05:
I can't find Serial.h indeed. Is there any other method to use the serial port (UART) on my stm32f446re?

0xc0170, 2021-01-05:
@hannbusann please update your first comment - it should keep the issue template as we parse it automatically to triage it further. I'll close this until it's fixed.

To answer the above, `Serial.h` was removed, use BufferedSerial: https://os.mbed.com/docs/mbed-os/v6.6/apis/serial-uart-apis.html

closed 2021-01-05 by 0xc0170