ARMmbed/mbed-os #14119: Dual Bank Flash Unable to Reboot on Hard Fault

lauleehong, 2021-01-06:
### Description of defect
- Using dual-bank flash in STM32F767ZI, crash and hard faults gets stuck.
- Tried both using ARMC 6.14 and GCC_ARM 10 compilers

GCC-ARM:
[image]

ARMC
[image]

#### Target(s) affected by this defect ?
STM32F767ZI

#### Toolchain(s) (name and version) displaying this defect ?
ARMC 6.14 and GCC_ARM 10

#### What version of Mbed-os are you using (tag or sha) ?
mbed-os 6.5.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

#### How is this defect reproduced ? 
- Set the F767ZI to dual bank flash mode using the STLink tool.
- Set mbed_app.json to reboot
`           "platform.error-reboot-max"                 : 1000,
            "platform.fatal-error-auto-reboot-enabled"  : true,`
- Flash application which invokes a hard fault
- Application expected to reboot automatically, but does not.
- Same hardfault invoked will auto reboot in mbed-os 5.14.2. (single bank flash)

jeromecoutant, 2021-01-06:
Hi

Please check https://os.mbed.com/teams/ST/wiki/How-to-enable-flash-dual-bank

Maybe you didn't enable FLASH_DUAL_BANK in mbed_app.json ?

Jerome

lauleehong, 2021-01-06:
@jeromecoutant 
`            "target.flash_dual_bank"                    : 1,`
was set

lauleehong, 2021-01-06:
Dual bank is working (not wrongly set), as PDMC 4.7.0 is functioning and was able to FOTA to bank 2.

lauleehong, 2021-01-06:
My initial thought was the dual bank flash, but upon further investigation using a blinky project.
1. blinky with mbed 6.5.0 was able to reset with dual bank flash
2. blinky with mbed 6.5.0 and bootloader was not able to reset with dual bank flash

lauleehong, 2021-01-06:
Blinky's mbed_app.json with bootloader
```
{
    "macros": ["MBED_HEAP_STATS_ENABLED=0",
                "MBED_STACK_STATS_ENABLED=0",
                "MBED_MEM_TRACING_ENABLED=0",
                "OS_THREAD_LIBSPACE_NUM=8",
                "ARM_UC_USE_PAL_BLOCKDEVICE=1",
                "MBED_FLASH_BANK_SIZE=MBED_ROM_SIZE/2"],
    "config": {},
    "target_overrides": {
        "*": {
            "target.features_add"                       : ["STORAGE", "BOOTLOADER"],
            "mbed-trace.enable"                         : 1,
            "mbed-trace.max-level"                      : "TRACE_LEVEL_INFO",
            "target.components_add"                     : ["FLASHIAP"],
            "target.printf_lib"                         : "std",
            "storage.storage_type"                      : "TDB_INTERNAL",
            "platform.stdio-baud-rate"                  : 115200,
            "platform.error-reboot-max"                 : 1000,
            "platform.fatal-error-auto-reboot-enabled"  : true,
            "platform.crash-capture-enabled"            : true,
            "target.OUTPUT_EXT"                         : "bin",
            "target.flash_dual_bank"                    : 1
        },
        "NUCLEO_F767ZI": {
            "target.macros_add"                         : ["DEVICE_UID_ADDR=0x1FF0F420"],
            "target.macros_remove"                      : ["MBEDTLS_CONFIG_HW_SUPPORT"],
            "target.bootloader_img"                     : "bootloader/mbed-bootloader-internal-flash-dual-bank-no-rot-f767-4.1.3.bin",
            "target.flash_dual_bank"                    : 1,
            "target.header_offset"                      : "0x8000",
            "target.app_offset"                         : "0x8400",
            "target.restrict_size"                      : "0xF7C00",
            "storage.storage_type"                      : "TDB_INTERNAL",
            "storage_tdb_internal.internal_base_address": "(MBED_ROM_START + MBED_FLASH_BANK_SIZE)",
            "storage_tdb_internal.internal_size"        : "(16*4*1024)"
        }
    }
}
```

lauleehong, 2021-01-07:
I've resolved the issue. This has to do with the mbed-bootloader using a small lib resulting in the compiler over-optimizing, and therefore causing the crash RAM to be corrupted. I've read out the crash crc, and indeed they do not match.
After switching to standard c lib, the behavior for dual bank flash using mbed-bootloader is as expected during self-injected hard faults - reboot automatically.

Fix:
`"target.c_lib"                             : "small",`
should be
`"target.c_lib"                             : "std",`

closed 2021-01-07 by lauleehong

teetak01, 2021-01-07:
Thanks @lauleehong for the detailed investigations!

We will check if the mbed-bootloader configurations should be fixed as proposed.